zephyrproject-rtos/zephyr #29308: GPIO bit banging i2c init before gpio clock init in stm32f401 plantform,cause same gpio can work.

CassiniaF, 2020-10-18:
I enable GPIOE 7 as bit banging i2s scl ,but i find it will first init 
1 gpio_config(context->scl_gpio, config->scl_pin,config->scl_flags | GPIO_OUTPUT_HIGH);
then init gpio_stm32_init.
cause GPIOE7 can't work.

CassiniaF, 2020-10-19:
I fixed it ,
let gpio driver init level to more earlier
gpio_stm32.c:set  level to PRE_KERNEL_1

	DEVICE_AND_API_INIT(gpio_stm32_## __suffix,			       \
			    __name,					       \
			    gpio_stm32_init,				       \
			    &gpio_stm32_data_## __suffix,		       \
			    &gpio_stm32_cfg_## __suffix,		       \
			    PRE_KERNEL_1,				       \  
			    CONFIG_KERNEL_INIT_PRIORITY_DEFAULT,	       \
			    &gpio_stm32_driver)

erwango, 2020-10-19:
@CassiniaF we've made some changes around this part lately. Can you provide the sha1 on which you're seeing this ?

CassiniaF, 2020-10-21:
This code you can find it at "gpio_stm32.c"

erwango, 2020-10-21:
> This code you can find it at "gpio_stm32.c"
> zephyr/drivers/gpio/gpio_stm32.c

I think I got this.
But as I said, we've made some changes recently on this file and the issue you're seeing may be linked with these changes. So I'd need to know what code version you're using to know if this is before or after the changes. This is why I'm asking the SHA1 of the latest zephyr commit you're using.
Can you run `git describe` command and provide the return value ?

CassiniaF, 2020-10-21:
zephyr-v2.4.0-557-gdc37f988e0

erwango, 2020-10-22:
@CassiniaF T be sure I understand, which I2C driver are you using ? `drivers/i2c/i2c_gpio.c` ?

CassiniaF, 2020-10-22:
> @CassiniaF To be sure I understand the use case, which I2C driver are you using ? `drivers/i2c/i2c_gpio.c` ?

Yes   drivers/i2c/i2c_gpio.c

CassiniaF, 2020-10-22:
"drivers/gpio/gpio_stm32.c"  should init before  "drivers/i2c/i2c_gpio.c" .

erwango, 2020-10-22:
> "drivers/gpio/gpio_stm32.c" should init before "drivers/i2c/i2c_gpio.c" .

Or: "drivers/i2c/i2c_gpio.c" should init after "drivers/gpio/gpio_stm32.c".
Point is that most of the gpio drivers use the same `POST_KERNEL` init level.
So either we should fix all gpio drivers, either i2c_gpio is the one to be fixed.

I think this is a good question for the i2c gpio maintainers. @pabigot, @mnkp, any advice here ?

pabigot, 2020-10-22:
In the end game yes, there needs to be a dependency.  #26616 will get us part-way there.

It's getting way to complex to keep doing this mix of hard-coded priorities with Kconfig overrides.

Many SoC GPIO drivers could initialize pre-kernel, since they should not have dependencies on any kernel capabilities (to operate from an ISR, which the generally must, they can't be using semaphores for mutex).

But I'd look favorably on a PR to move `i2c_gpio` to init at `POST_KERNEL` instead of `PRE_KERNEL_2`.  I have no idea why it's different other than it's always been that way.  That may be enough to fix this particular problem.

@CassiniaF Could you please test whether i2c_gpio init at `POST_KERNEL`, without changing `gpio_stm32`, fixes your problem?  If so, please submit a PR with that fix.  If you can't do that let me know and I'll give it a shot.

CassiniaF, 2020-10-23:
> In the end game yes, there needs to be a dependency. #26616 will get us part-way there.
>
> It's getting way to complex to keep doing this mix of hard-coded priorities with Kconfig overrides.
>
> Many SoC [...]

 change i2c_gpio init at POST_KERNEL, without changing gpio_stm32, It FIX my issue.

CassiniaF, 2020-10-24:
Request a PR is not easy，email issue need me set my email public in my github Settings .
I tried and explored for a long time before I found this problem. I suggested writing it in the guideline

pabigot, 2020-10-24:
@CassiniaF Sorry for the difficulties; most people fork the repository and use git from the command line or an application on their local computer.  Then it's the local git installation that needs the submitter email address.  (Also with a forked repository you can push updates to your PR, so you don't have to open a new one each time.)

Thanks for your contribution; it looks good now.

PR #29515 (2020-10-26): drivers/i2c: i2c_gpio:init at POST_KERNEL
This is to make the gpio-i2c initialization occur after the GPIO pin
clock initialization.

Fixes #29308
closed 2020-10-26 by nashif