zephyrproject-rtos/zephyr #29309: ADC1 doesn't read correctly on STM32F7

disposedtrolley, 2020-10-18:
**Describe the bug**
The ADC_1 peripheral doesn't seem to work correctly on the STM32F7 series. I've tried compiling a test project following the test code supplied in the [PR](https://github.com/zephyrproject-rtos/zephyr/pull/13447#issuecomment-464342445) which introduced ADC support for STM32, but I get garbage readings on my Nucleo F767ZI board. The same code works fine on my Nucleo F401RE.

My settings:

- F767ZI
  - ADC_1
  - Channel 12
- F401RE
  - ADC_1
  - Channel 0

**To Reproduce**
Steps to reproduce the behavior:
1. Build a minimal project according to the [test code](https://github.com/zephyrproject-rtos/zephyr/pull/13447#issuecomment-464342445) targeting the Nucleo F676ZI board. Use the following channel config:
```
static const struct adc_channel_cfg adc_cfg = {
    .gain = ADC_GAIN_1,
    .reference = ADC_REF_INTERNAL,
    .channel_id = 12
};
```
  - If gain and reference aren't provided when building for the F7, it results in errors when executing `adc_channel_setup()`.
2. Connect something to Pin 7 of CN10 (PB1).
3. Observe that garbage readings are printed to the console.

**Expected behavior**
The ADC should output similar readings on the F7 series as the F4.

**Impact**
It was painful to debug whether this was an issue with my code, a hardware problem, or a Zephyr issue. Since the ADC implementation was only tested on some F4 boards, perhaps compatibility with other STM32 series should be removed from the docs.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain: arm-none-eabi-gcc
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

disposedtrolley, 2020-10-18:
Okay I think I know what the issue is; the pinmux definitions for the STM32 Nucleo boards only map PA0 to ADC1. None of the other pins that are supported by the ADCs are mapped.

I'll see if I can open up a PR soon to fix this, at least for the F767ZI!

disposedtrolley, 2020-10-18:
Nevermind! I don't think any changes are required -- I just needed to create a devicetree overlay file and override the `&adc1` definition so it includes the GPIO pins I'm using, i.e.

```
&adc1 {
	pinctrl-0 = <&adc1_in0_pa0 &adc1_in9_pb1>;
	status = "okay";
};
```

which adds `&adc1_in9_pb1` so I can connect something to PB1 and read from channel 9 of ADC1.

erwango, 2020-10-19:
@disposedtrolley As I understand, everything work fine then ?

disposedtrolley, 2020-10-19:
Yep all good @erwango

Do you think it's worthwhile to document the pinmuxing somewhere? Would there be a suitable section in the docs to cover this?

erwango, 2020-10-22:
> Do you think it's worthwhile to document the pinmuxing somewhere? Would there be a suitable section in the docs to cover this?

@disposedtrolley This is indeed required. My view is that it should belong to upcoming DT bindings doc section: https://github.com/zephyrproject-rtos/zephyr/pull/29374

disposedtrolley, 2020-10-22:
> This is indeed required. My view is that it should belong to upcoming DT bindings doc section: #29374

@erwango Fantastic. I'll keep an eye out for the PR to be merged and see how the pinmux definitions look. I'll try adding documentation as necessary.

erwango, 2020-10-22:
@disposedtrolley Txs for feedback. Feel free to close this point if we're done.

closed 2020-10-22 by disposedtrolley