zephyrproject-rtos/zephyr #29310: ble central Repeat read and write to three peripherals error USAGE FAULT

mfinmuch, 2020-10-18:
Currently I can connect one central to three peripherals
After the connection is established, the corresponding uuid and handle are also found
I use these two functions to read/write
`bt_gatt_write(bt_conn_lookup_index(conn_index), &write_params);` 
 `bt_gatt_read(bt_conn_lookup_index(conn_index), &read_params);`
conn_idex is the value I entered by myself, it seems I want to connect to the number (if I want to write/read to three peripherals in sequence, fill in 0,1,2 in sequence)

But when central sent read/write to peripheral and ran for one round, the following error occurred when it was about to run the second round
\`\`\`
[00:00:06.042,266] <err> os: ***** USAGE FAULT *****
[00:00:06.042,297] <err> os: Illegal use of the EPSR
[00:00:06.042,297] <err> os: r0/a1: 0x20000218 r1/a2: 0x00000001 r2/a3: 0x200016fc
[00:00:06.042,297] <err> os: r3/a4: 0x00000000 r12/ip: 0x0000cfe9 r14/lr: 0x0000de87
[00:00:06.042,297] <err> os: xpsr: 0x20000000
[00:00:06.042,297] <err> os: Faulting instruction address (r15/pc): 0x00000000
[00:00:06.042,327] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:06.042,327] <err> os: Current thread: 0x200013f0 (unknown)
[00:00:06.780,883] <err> os: Halting system
\`\`\`
How can I remove this error?
Is there a problem with the `bt_conn_lookup_index `I quoted in bt_gatt_write/read?
What to do to make one central communicate with multiple peripherals?

Environment:

> Zephyr 2.4.0

> zephyr-sdk-0.11.3

> nrf52840

joerchan, 2020-10-20:
Should check the return value of bt_conn_lookup_index.
I would recommend to enable CONFIG_ASSERT, there is probably an assert before the usage fault.

carlescufi, 2020-11-05:
@mfinmuch any updates?

carlescufi, 2020-12-03:
Closing because of inactivity, please reopen if this is still an issue.

closed 2020-12-03 by carlescufi