zephyrproject-rtos/zephyr #29315: twr_kv58f220m: all application build failure

hakehuang, 2020-10-19:
**Describe the bug**
missing fsl_commom.h

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=twr_kv58f220m ..
3. make
4. See error

**Expected behavior**
build pass

**Impact**
twr_kv58f220m support

**Logs and console output**

```
/build/src/workspace/twr_kv58f220m_3_master

Including boilerplate (Zephyr base): /build/src/workspace/twr_kv58f220m_3_master/zephyr/cmake/app/boilerplate.cmake

-- Application: /build/src/workspace/twr_kv58f220m_3_master/zephyr/samples/hello_world

-- Zephyr version: 2.4.99 (/build/src/workspace/twr_kv58f220m_3_master/zephyr)

-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 

-- Found west (found suitable version "0.7.2", minimum required is "0.7.1")

-- git describe failed: fatal: No names found, cannot describe anything.;

   BUILD_VERSION is left undefined

-- Board: twr_kv58f220m

-- Cache files will be written to: /root/.cache/zephyr

-- Found toolchain: zephyr (/opt/zephyr-sdk)

-- Found dtc: /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")

-- Found BOARD.dts: /build/src/workspace/twr_kv58f220m_3_master/zephyr/boards/arm/twr_kv58f220m/twr_kv58f220m.dts

-- Generated zephyr.dts: /build/src/workspace/twr_kv58f220m_3_master/zephyr/samples/hello_world/build/zephyr/zephyr.dts

-- Generated devicetree_unfixed.h: /build/src/workspace/twr_kv58f220m_3_master/zephyr/samples/hello_world/build/zephyr/include/generated/devicetree_unfixed.h

Parsing /build/src/workspace/twr_kv58f220m_3_master/zephyr/Kconfig

Loaded configuration '/build/src/workspace/twr_kv58f220m_3_master/zephyr/boards/arm/twr_kv58f220m/twr_kv58f220m_defconfig'

Merged configuration '/build/src/workspace/twr_kv58f220m_3_master/zephyr/samples/hello_world/prj.conf'

Configuration saved to '/build/src/workspace/twr_kv58f220m_3_master/zephyr/samples/hello_world/build/zephyr/.config'

Kconfig header saved to '/build/src/workspace/twr_kv58f220m_3_master/zephyr/samples/hello_world/build/zephyr/include/generated/autoconf.h'

-- The C compiler identification is GNU 9.2.0

-- The CXX compiler identification is GNU 9.2.0

-- The ASM compiler identification is GNU

-- Found assembler: /opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc

-- Configuring done

-- Generating done

-- Build files have been written to: /build/src/workspace/twr_kv58f220m_3_master/zephyr/samples/hello_world/build

Scanning dependencies of target parse_syscalls_target

[  1%] Generating misc/generated/syscalls_subdirs.trigger

[  1%] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json

[  2%] Built target parse_syscalls_target

Scanning dependencies of target kobj_types_h_target

[  3%] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h

[  3%] Built target kobj_types_h_target

Scanning dependencies of target syscall_list_h_target

[  4%] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h

[  6%] Built target syscall_list_h_target

Scanning dependencies of target driver_validation_h_target

[  7%] Generating include/generated/driver-validation.h

[  9%] Built target driver_validation_h_target

Scanning dependencies of target offsets

[ 10%] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj

In file included from /build/src/workspace/twr_kv58f220m_3_master/modules/hal/nxp/mcux/devices/MKV58F24/fsl_device_registers.h:23,

                 from /build/src/workspace/twr_kv58f220m_3_master/modules/hal/nxp/mcux/drivers/kinetis/./fsl_common.h:31,

                 from /build/src/workspace/twr_kv58f220m_3_master/zephyr/soc/arm/nxp_kinetis/kv5x/soc.h:18,

                 from /build/src/workspace/twr_kv58f220m_3_master/zephyr/include/arch/arm/aarch32/cortex_m/cmsis.h:17,

                 from /build/src/workspace/twr_kv58f220m_3_master/zephyr/include/arch/arm/aarch32/cortex_m/mpu/arm_mpu_v7m.h:10,

                 from /build/src/workspace/twr_kv58f220m_3_master/zephyr/include/arch/arm/aarch32/cortex_m/mpu/arm_mpu.h:13,

                 from /build/src/workspace/twr_kv58f220m_3_master/zephyr/include/arch/arm/aarch32/arch.h:186,

                 from /build/src/workspace/twr_kv58f220m_3_master/zephyr/include/arch/cpu.h:19,

                 from /build/src/workspace/twr_kv58f220m_3_master/zephyr/include/kernel_includes.h:38,

                 from /build/src/workspace/twr_kv58f220m_3_master/zephyr/include/kernel.h:17,

                 from /build/src/workspace/twr_kv58f220m_3_master/zephyr/arch/arm/core/offsets/offsets_aarch32.c:28,

                 from /build/src/workspace/twr_kv58f220m_3_master/zephyr/arch/arm/core/offsets/offsets.c:12:

/build/src/workspace/twr_kv58f220m_3_master/modules/hal/nxp/mcux/devices/MKV58F24/MKV58F24.h:272:10: fatal error: core_cm7.h: No such file or directory

  272 | #include "core_cm7.h"                  /* Core Peripheral Access Layer */

      |          ^~~~~~~~~~~~

compilation terminated.

zephyr/CMakeFiles/offsets.dir/build.make:81: recipe for target 'zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj' failed

make[2]: *** [zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj] Error 1

CMakeFiles/Makefile2:2554: recipe for target 'zephyr/CMakeFiles/offsets.dir/all' failed

make[1]: *** [zephyr/CMakeFiles/offsets.dir/all] Error 2

make: *** [all] Error 2

Makefile:102: recipe for target 'all' failed

script returned exit code 2

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: https://github.com/zephyrproject-rtos/zephyr/commit/260b327d2f27aacb123a099374c6b4c2277164b4

hakehuang, 2020-10-19:
@MaureenHelm

nashif, 2020-10-19:
works for me, do you have west updated and environment setup correctly? Something like this would have been caught in CI..

MaureenHelm, 2020-10-19:
Works for me too. Tested on 260b327d2f27aacb123a099374c6b4c2277164b4 and ad8caa65e8578cfbbe51f233192099c2190d9c7f

hakehuang, 2020-10-20:
sorry, I find this issue is the west update failure, after check the log, it show, it is a network issue at my office.

```
--- hal_nxp: initializing

Initialized empty Git repository in /build/src/workspace/twr_kv58f220m_master/modules/hal/nxp/.git/

--- hal_nxp: fetching, need revision 68bbdbdec1153355b14386d8f095058a496f66f4

fatal: unable to access 'https://github.com/zephyrproject-rtos/hal_nxp/': Failed to connect to github.com port 443: Connection timed out
```

closed 2020-10-20 by hakehuang