zephyrproject-rtos/zephyr #29327: mec15xxevb_assy6853:region `SRAM' overflowed during build

maksimmasalski, 2020-10-19:
**Describe the bug**
Can't build test tests/portability/cmsis_rtos_v2/ using west or sanitycheck I receive error that SRAM overflowed.

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -p -b mec15xxevb_assy6853 tests/portability/cmsis_rtos_v2/`
or 
2. `sanitycheck --hardware-map ~/mec1.file --device-testing -T tests/portability/cmsis_rtos_v2/ -vv`
4. See error

**Expected behavior**
Test should be build and passed for the Microchip.

**Impact**
Can't execute that test

**Logs and console output**
For west  Output:
```
-- west build: making build dir /home/ztest/work/zephyrproject/zephyr/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/ztest/work/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/ztest/work/zephyrproject/zephyr/tests/portability/cmsis_rtos_v2
-- Zephyr version: 2.4.99 (/home/ztest/work/zephyrproject/zephyr)
-- Found Python3: /usr/bin/python3.6 (found suitable exact version "3.6.9") found components:  Interpreter
-- Found west (found suitable version "0.7.2", minimum required is "0.7.1")
-- Board: mec15xxevb_assy6853
-- Cache files will be written to: /home/ztest/.cache/zephyr
-- Found toolchain: zephyr (/home/ztest/zephyr-sdk-0.11.4)
-- Found dtc: /home/ztest/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/ztest/work/zephyrproject/zephyr/boards/arm/mec15xxevb_assy6853/mec15xxevb_assy6853.dts
-- Generated zephyr.dts: /home/ztest/work/zephyrproject/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/ztest/work/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h
Parsing /home/ztest/work/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/ztest/work/zephyrproject/zephyr/boards/arm/mec15xxevb_assy6853/mec15xxevb_assy6853_defconfig'
Merged configuration '/home/ztest/work/zephyrproject/zephyr/tests/portability/cmsis_rtos_v2/prj.conf'
Configuration saved to '/home/ztest/work/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/ztest/work/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'

warning: MAX_THREAD_BYTES (defined at arch/Kconfig:273) was assigned the value '4' but got the value
''. Check these unsatisfied dependencies: USERSPACE (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_MAX_THREAD_BYTES.html and/or look up
MAX_THREAD_BYTES in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

-- The C compiler identification is GNU 9.2.0
-- The CXX compiler identification is GNU 9.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ztest/zephyr-sdk-0.11.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ztest/work/zephyrproject/zephyr/build
-- west build: building application
[1/160] Preparing syscall dependency handling

[155/160] Linking C executable zephyr/zephyr_prebuilt.elf
FAILED: zephyr/zephyr_prebuilt.elf
: && ccache /home/ztest/zephyr-sdk-0.11.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc    zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj  -o zephyr/zephyr_prebuilt.elf  -Wl,-T zephyr/linker.cmd -Wl,-Map=/home/ztest/work/zephyrproject/zephyr/build/zephyr/zephyr_prebuilt.map -Wl,--whole-archive app/libapp.a zephyr/libzephyr.a zephyr/arch/common/libarch__common.a zephyr/arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a zephyr/arch/arch/arm/core/aarch32/cortex_m/libarch__arm__core__aarch32__cortex_m.a zephyr/lib/libc/minimal/liblib__libc__minimal.a zephyr/lib/posix/liblib__posix.a zephyr/lib/cmsis_rtos_v2/liblib__cmsis_rtos_v2.a zephyr/boards/arm/mec15xxevb_assy6853/libboards__arm__mec15xxevb_assy6853.a zephyr/subsys/testsuite/ztest/libsubsys__testsuite__ztest.a zephyr/drivers/adc/libdrivers__adc.a zephyr/drivers/gpio/libdrivers__gpio.a zephyr/drivers/i2c/libdrivers__i2c.a zephyr/drivers/pwm/libdrivers__pwm.a zephyr/drivers/sensor/mchp_tach_xec/libdrivers__sensor__mchp_tach_xec.a zephyr/drivers/espi/libdrivers__espi.a zephyr/drivers/ps2/libdrivers__ps2.a zephyr/drivers/serial/libdrivers__serial.a -Wl,--no-whole-archive zephyr/kernel/libkernel.a zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj -L"/home/ztest/zephyr-sdk-0.11.4/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/thumb/v7e-m/nofp" -L/home/ztest/work/zephyrproject/zephyr/build/zephyr -lgcc -Wl,--print-memory-usage zephyr/arch/common/libisr_tables.a -mcpu=cortex-m4 -mthumb -mabi=aapcs -Wl,--gc-sections -Wl,--build-id=none -Wl,--sort-common=descending -Wl,--sort-section=alignment -Wl,-u,_OffsetAbsSyms -Wl,-u,_ConfigAbsSyms -nostdlib -static -no-pie -Wl,-X -Wl,-N -Wl,--orphan-handling=warn && :
Memory region         Used Size  Region Size  %age Used
           FLASH:       75916 B       224 KB     33.10%
            SRAM:       40036 B        32 KB    122.18%
        IDT_LIST:         216 B         2 KB     10.55/home/ztest/zephyr-sdk-0.11.4/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: zephyr/zephyr_prebuilt.elf section `noinit' will not fit in region `SRAM'
/home/ztest/zephyr-sdk-0.11.4/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: section .intList VMA [0000000000120000,00000000001200d7] overlaps section noinit VMA [000000000011aee8,0000000000121c63]
/home/ztest/zephyr-sdk-0.11.4/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/9.2.0/../../../../arm-zephyr-eabi/bin/ld: region `SRAM' overflowed by 7268 bytes
collect2: error: ld returned 1 exit status
%
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/ztest/bin/cmake/cmake-3.13.1-Linux-x86_64/bin/cmake --build /home/ztest/work/zephyrproject/zephyr/build
```

For sanitycheck  Output:
```
Device testing on:

| Platform            | ID       | Serial device   |
|---------------------|----------|-----------------|
| mec15xxevb_assy6853 | FT9FPJYP | /dev/ttyUSB2    |

INFO    - Adding tasks to the queue...
DEBUG   - Running cmake on /home/ztest/work/zephyrproject/zephyr/tests/portability/cmsis_rtos_v2 for mec15xxevb_assy6853
DEBUG   - DONE FEEDING
DEBUG   - Calling cmake with arguments: ['-B/home/ztest/work/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/portability/cmsis_rtos_v2/portability.cmsis_rtos_v2', '-S/home/ztest/work/zephyrproject/zephyr/tests/portability/cmsis_rtos_v2', '-DEXTRA_CFLAGS="-Werror"', '-DEXTRA_AFLAGS="-Wa,--fatal-warnings', '-DEXTRA_LDFLAGS="-Wl,--fatal-warnings"', '-GUnix Makefiles', '-DBOARD=mec15xxevb_assy6853']
DEBUG   - Finished building /home/ztest/work/zephyrproject/zephyr/tests/portability/cmsis_rtos_v2 for mec15xxevb_assy6853
DEBUG   - build test: mec15xxevb_assy6853/tests/portability/cmsis_rtos_v2/portability.cmsis_rtos_v2
DEBUG   - Building /home/ztest/work/zephyrproject/zephyr/tests/portability/cmsis_rtos_v2 for mec15xxevb_assy6853
DEBUG   - Test skipped due to SRAM Overflow
INFO    - 1/1 mec15xxevb_assy6853       tests/portability/cmsis_rtos_v2/portability.cmsis_rtos_v2 SKIPPED (SRAM overflow)

DEBUG   - running footprint_reports
INFO    - 0 of 0 tests passed (0.00%), 0 failed, 1 skipped with 0 warnings in 4.45 seconds
INFO    - In total 0 test cases were executed on 1 out of total 292 platforms (0.34%)
INFO    - 0 tests executed on platforms, 0 tests were only built.
```

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 18.04.4 LTS
 - Toolchain Zephyr SDK 0.11.4
 - Commit SHA or Version used 53cc090c39a538289eac538cb7fb0dc38a1f8d6f

maksimmasalski, 2020-10-19:
@scottwcpg Do you have any thoughts about that?

nashif, 2020-10-19:
this is not a bug, the test/sample just does not fit on the device.

closed 2020-10-19 by nashif

maksimmasalski, 2020-10-27:
Track issue using #29412 That is a root cause