zephyrproject-rtos/zephyr #29329: tests: kernel.workqueue.api tests fail on nrf platforms

PerMac, 2020-10-19:
**Describe the bug**
Tests "work_submit_to_multipleq" and "work_resubmit_nowait_to_queue " from "tests/kernel/workq/work_queue_api" fail on nrf platforms (nrf52840dk_nrf52840, nrf52dk_nrf52832 and nrf9160dk_nrf9160)

**To Reproduce**
Steps to reproduce the behavior:
1. go to zephyr dir
2. call `./scripts/sanitycheck --hardware-map map.yml --device-testing -T tests/kernel/workq/work_queue_api/`
3. See error

**Expected behavior**
The tests pass

**Impact**
Not clear

**Logs and console output**

```
 START - test_work_submit_to_multipleq Assertion failed at CMAKE_SOURCE_DIR/src/main.c:277: twork_submit_multipleq: k_delayed_work_pending(&new_work) is false FAIL - test_work_submit_to_multipleq
 ```
and
```
START - test_work_resubmit_nowait_to_queue Assertion failed at CMAKE_SOURCE_DIR/src/main.c:357: twork_resubmit_nowait: k_work_pending(&delayed_work_sleepy.work) is true FAIL - test_work_resubmit_nowait_to_queue 
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain Zephyr SDK v0.11.3
 - Zephyr version: zephyr-v2.4.0-685-g7e64fba9ed

maksimmasalski, 2020-10-19:
#29328 Same thing

nashif, 2020-10-19:
bisection leads to bd76994dc21b3b6a7a2d6a02413c20cab82762ce, FYI @pabigot

nashif, 2020-10-19:
does not seem to be nrf specific.

pabigot, 2020-10-19:
I've extracted the fixes for these that were in #28891 since that's stalled.

PR #29335 (2020-11-02): tests: kernel: work_queue_api: relax test to eliminate racy checks
We can't control ticks accurately enough to detect the transition
between on a queue and being handled, so relax the checks to make
things pass.

Fixes #29329
closed 2020-11-02 by nashif