zephyrproject-rtos/zephyr #29346: west boards doesn't display the arcitecture.

tejlmand, 2020-10-20:
**Describe the bug**
Running `west boards` provides a list of all boards in one big list, like:
```
$ west boards
em_starterkit
em_starterkit_em11d
em_starterkit_em7d
em_starterkit_em7d_v22
emsdp
emsdp_em4
emsdp_em5d
emsdp_em6
emsdp_em7d
emsdp_em7d_esp
emsdp_em9d
hsdk_2cores
hsdk
iotdk
nsim_em7d_v22
nsim_em
nsim_hs
nsim_hs_smp
nsim_sem
nsim_sem_mpu_stack_guard
qemu_arc_em
qemu_arc_hs
96b_aerocore2
96b_argonkey
....
```

whereas running `cmake -P cmake/boards.cmake` (which is used by `west boards`, https://github.com/zephyrproject-rtos/zephyr/blob/master/scripts/west_commands/boards.py#L58) gives a nice arch organized presentation of the boards. 
```
$ cmake -P cmake/boards.cmake 
arc:
  em_starterkit
  em_starterkit_em11d
  em_starterkit_em7d
  em_starterkit_em7d_v22
  emsdp
  emsdp_em4
  emsdp_em5d
  emsdp_em6
  emsdp_em7d
  emsdp_em7d_esp
  emsdp_em9d
  hsdk_2cores
  ...
arm:
  96b_aerocore2
  96b_argonkey
  96b_avenger96
  96b_carbon
  96b_carbon_nrf51
  ...
nios2:
  altera_max10
  qemu_nios2
posix:
  native_posix_64
  native_posix
  nrf52_bsim
riscv:
  hifive1
  hifive1_revb
  litex_vexriscv
  m2gl025_miv
  qemu_riscv32
  qemu_riscv64
  rv32m1_vega_ri5cy
  rv32m1_vega_zero_riscy
x86:
  acrn
  minnowboard
  qemu_x86_64
  qemu_x86_coverage
....
```

**To Reproduce**
`west boards` vs `cmake -P cmake/boards.cmake`

**Expected behavior**
A board list similar to `cmake -P cmake/boards.cmake`, as that presents a nice overview for users.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain N/A
 - zephyr/master

**Additional context**
Add any other context about the problem here.

nashif, 2020-10-20:
```west boards -f "{arch} - {name}"```

tejlmand, 2020-10-20:
> `west boards -f "{arch} - {name}"`

Nice, didn't know that ;-) (although output it's still a little different)
Question still is, what would we like to be default, and is there a special reason why we strip the grouping per default, now that we have it ?

Which were the main reason I raised this.

nashif, 2020-10-24:
rather than turn it to a question, I would prefer to close this as a non-bug.

closed 2020-10-24 by nashif

tejlmand, 2020-10-26:
@nashif we already have a nice formatted list, why remove that formatting just, an then ask users to re-add it in slightly different way ?