zephyrproject-rtos/zephyr #29363: shell: inability to print 64-bit integers with newlib support

e-rk, 2020-10-20:
**Describe the bug**
When `CONFIG_NEWLIB_LIBC=y` is used, the shell printf is unable to substitute the `%lu` specifier. 

**To Reproduce**
Steps to reproduce the behavior:
1. `cd samples/subsys/shell/shell_module`
2. Add `CONFIG_NEWLIB_LIBC=y` to `prj.conf`
3. Add the following lines to any command handler:

        uint64_t val = 51;
        shell_print(shell, "Test: %" PRIu64, val);

2. `west build -b nrf52840dk_nrf52840 -p && west flash`
3. Execute the command

        uart:~$ demo ping
        pong
        Test: %lu

4. Observe, that `%lu` is displayed instead of a number.

**Expected behavior**
The specifier is correctly substituted with a 64-bit value.

**Impact**
Inability to print 64-bit values with the shell.

**Logs and console output**
Output shown above.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain `gcc-arm-none-eabi-9-2019-q4-major`
 - Commit SHA or Version used: `4d0750920b52e0ce8a3e188d08a1769ff7ec527b`

**Additional context**
The value used to be printed correctly with values less than 2^32. `git bisect` points to:
```
e812ee6c213143fdc94c20de1b8d26db38cd0cc0 is the first bad commit
commit e812ee6c213143fdc94c20de1b8d26db38cd0cc0
Author: Peter Bigot <peter.bigot@nordicsemi.no>
Date:   Tue Aug 18 14:03:13 2020 -0500

    shell: support floating point output with newlib
    
    shell_fprintf requires that formatted output be emitted with a
    putchar()-like output function.  Newlib does not provide such a
    capability.  Zephyr provides two solutions: z_prf() which is part of
    minimal libc and handles floating point formatting, and z_vprintk()
    which is core and does not support floating point.
    
    Move z_prf() out of minimal libc into the core lib area, and use it
    unconditionally in the shell.
    
    Signed-off-by: Peter Bigot <peter.bigot@nordicsemi.no>

:040000 040000 e41e21bd573bb9bb8de892ef95003c599f7585a8 c1dca9a4609ed2b63ffc30b37e93d0d2519bc8e2 M	lib
:040000 040000 7e472664932e6743c012c193c2e2e3810b7d7051 3aa84564ef4ab80c739c9c8a3ee836591c642fa6 M	subsys
```
It appears, that the `z_vprintk` used previously could substitute `%lu`, but only up to 2^32-1. The `z_prf` also has this ability, but it is blocked behind `CONFIG_MINIMAL_LIBC_LL_PRINTF` which can't be enabled due to `CONFIG_NEWLIB_LIBC=y`.

pabigot, 2020-10-20:
This was going to be fixed by #29171 but that got shot down.  We don't have a way forward until the discussion in https://github.com/zephyrproject-rtos/zephyr/pull/29203#issuecomment-710072323 reaches resolution.

pabigot, 2020-12-20:
Should be closed by #29876

closed 2020-12-20 by pabigot