zephyrproject-rtos/zephyr #29369: Devicetree binding/description for pinmux on GPIO controller based cfg SoCs

galak, 2020-10-20:
This issue is to discuss ideas on a generic / template for describing pinmux/cfg information for SoCs that handle pincfg as part of the GPIO block.  Examples of such SoCs are STM, NXP, etc.

In such systems typically the following information is related to a pin:
* port (which GPIO block the pin is on - (GPIO A, GPIO B, GPIO C, PORT A, PORT B, PORT C, PIO 0, PIO 1, etc.)
* pin (typically a 0..31) number for the pin on the port
* function - how the pin is configured (used as GPIO, analog, I2C_SCL, etc)
* attributes/flags - additional params of the pin (pull-up/pull-down, speed/slew-rate, etc).

some how all this information needs to be encoded in the devicetree.

galak, 2020-10-20:
PORT:
Option 1 - encode via hierarchy - place pincfg nodes under the port device that contains the registers to configure that pin.

```
                porta: pinmux@40049000 {
                        compatible = "nxp,kinetis-pinmux";
                        reg = < 0x40049000 0xd0 >;
                        clocks = < &sim 0x2 0x1038 0x9 >;
                        phandle = < 0x4 >;
                        pinctrl {
                        };
                };
```

Option 2 - place all pinctrl in one block and provide references per pin to the port (in example below encoding is via phandle reference, could be encoded in some other means)

```
pinctrl {
   foo {
      pin = <&porta 3 ... >;
  };
  bar {
      pin = <&portc 6 ... >;
  };    
};
```

galak, 2020-10-20:
**FLAGS/ATTRIBUTES:**

Utilize common properties as per:

https://git.kernel.org/pub/scm/linux/kernel/git/devicetree/devicetree-rebasing.git/tree/Bindings/pinctrl/pincfg-node.yaml

Than per SoC bindings can add/remove/modify additional flags/attributes as it makes sense.

erwango, 2020-10-20:
I'd add one assumption:
* The proposed template should be generation friendly 

Reason is: one of these mcu characteristics is the high number of possible signals that are made available.
To make them available in an exhaustive way, usage of generation engines is likely mandatory.

One open question:
Should we target compatibility with Linux bindings ?
Note that compatibility doesn't exclude to define a zephyr specific bindings. I think that DT API could help be compatible with both models.

galak, 2020-10-20:
> I'd add one assumption:
>
> * The proposed template should be generation friendly
>
> Reason is: one of these mcu characteristics is the high number of possible signals that are made available.
> To make [...]

My idea here is to define a common binding for SoCs that don't having any pre-existing DTS binding.  However, we should try to utilize concepts from the upstream bindings unless we feel strongly about differing from them.

For example, here are some different ways we could encode things:

example: Based on pinmux-node.yaml: tweak to make pin & function numbers and having explicit port reference
```
{
  port = <&portC>;
  pin = <3>;
  function = <4>; /* soc defined purpose */
    ... /* flags like bias-pull-down */
}
```

example - put all port/pin/function together:
{
   signal = <&portC 3 4>; /* port pin function>
    ... /* flags like bias-pull-down */
}

tannewt, 2020-10-20:
Could you point to two specific SoCs that have a different IO structure? Looking at the iMX RT 1010 it looks like it has an IOMUXC separate from a GPIO block. I'm not sure what difference you are pointing out in this issue.

galak, 2020-10-20:
> Could you point to two specific SoCs that have a different IO structure? Looking at the iMX RT 1010 it looks like it has an IOMUXC separate from a GPIO block. I'm not sure what difference you are pointing out in this issue.

If you look at a Silabs or Nordic SoC you'll see that the function selection for how a pin is used is done in the IP block.  For example in Nordic SOCs you'll see PSEL registers in the UART block that assign the port & pin.  So each IP block has a set of registers.  Compare that to a iMX RT in which all the function config is done in the IOMUXC (ie centralized).

jettr, 2020-10-20:
For the Port, I like option 2:
>Option 2 - place all pinctrl in one block and provide references per pin to the port (in example below encoding is via phandle reference, could be encoded in some other means)

Also 
>My idea here is to define a common binding for SoCs

I think this is a great idea. There are a very common list of things a pins can do, and we should definitely standardize this (maybe a common.yml definition). Chip specific functionality could extend the common definition.