zephyrproject-rtos/zephyr #29376: sanitycheck: "TypeError: 'NoneType' object is not iterable"

andrewboie, 2020-10-20:
```
$ scr -p qemu_x86_64 -p qemu_x86 -T tests/kernel/
Deleting output directory /dev/shm/sanity-out
INFO    - JOBS: 48
INFO    - Building initial testcase list...
INFO    - 214 test configurations selected, 32 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:   35/ 182  19%  skipped:   14, failed:    0
ERROR   - qemu_x86_64               tests/kernel/mutex/sys_mutex/system.mutex          FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_x86_64/tests/kernel/mutex/sys_mutex/system.mutex/handler.log
INFO    - Total complete:   41/ 182  22%  skipped:   14, failed:    1
ERROR   - qemu_x86                  tests/kernel/mutex/sys_mutex/system.mutex          FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/mutex/sys_mutex/system.mutex/handler.log
INFO    - Total complete:   94/ 182  51%  skipped:   15, failed:    2
ERROR   - qemu_x86_64               tests/kernel/mem_pool/mem_pool_concept/kernel.memory_pool.concept.legacy FAILED: Failed
ERROR   - see: /dev/shm/sanity-out/qemu_x86_64/tests/kernel/mem_pool/mem_pool_concept/kernel.memory_pool.concept.legacy/handler.log
INFO    - Total complete:   95/ 182  52%  skipped:   17, failed:    3ERROR   - 'qemu_x86/tests/kernel/interrupt/arch.interrupt' generated an exception:
ERROR   - Traceback (most recent call last):
ERROR   -   File "/home/apboie/projects/zephyr/zephyr/scripts/sanity_chk/sanitylib.py", line 3072, in execute
ERROR   -     data = future.result()
ERROR   -   File "/usr/lib/python3.8/concurrent/futures/_base.py", line 432, in result
ERROR   -     return self.__get_result()
ERROR   -   File "/usr/lib/python3.8/concurrent/futures/_base.py", line 388, in __get_result
ERROR   -     raise self._exception
ERROR   -   File "/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
ERROR   -     result = self.fn(*self.args, **self.kwargs)
ERROR   -   File "/home/apboie/projects/zephyr/zephyr/scripts/sanity_chk/sanitylib.py", line 2145, in process
ERROR   -     self.cleanup_artifacts()
ERROR   -   File "/home/apboie/projects/zephyr/zephyr/scripts/sanity_chk/sanitylib.py", line 2157, in cleanup_artifacts
ERROR   -     allow += additional_keep
ERROR   - TypeError: 'NoneType' object is not iterable
INFO    - Total complete:   97/ 182  53%  skipped:   19, failed:    3'qemu_x86/tests/kernel/interrupt/arch.interrupt' generated an exception: 'NoneType' object is not iterable
apboie@apboie-mobl4:~/projects/zephyr/zephyr (1) page-tables-overhaul /home/apboie/projects/zephyr/zephyr
$ scr -p qemu_x86_64 -p qemu_x86 -T tests/kernel/
Deleting output directory /dev/shm/sanity-out
INFO    - JOBS: 48
INFO    - Building initial testcase list...
INFO    - 214 test configurations selected, 32 configurations discarded due to filters.
INFO    - Adding tasks to the queue...
INFO    - Total complete:   20/ 182  10%  skipped:   14, failed:    0
ERROR   - qemu_x86_64               tests/kernel/lifo/lifo_usage/kernel.lifo.usage     FAILED: Failed
ERROR   - see: /dev/shm/sanity-out/qemu_x86_64/tests/kernel/lifo/lifo_usage/kernel.lifo.usage/handler.log
INFO    - Total complete:   41/ 182  22%  skipped:   14, failed:    1
ERROR   - qemu_x86_64               tests/kernel/mutex/sys_mutex/system.mutex          FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_x86_64/tests/kernel/mutex/sys_mutex/system.mutex/handler.log
INFO    - Total complete:   43/ 182  23%  skipped:   14, failed:    2
ERROR   - qemu_x86                  tests/kernel/mutex/sys_mutex/system.mutex          FAILED: Exited with 2
ERROR   - see: /dev/shm/sanity-out/qemu_x86/tests/kernel/mutex/sys_mutex/system.mutex/handler.log
INFO    - Total complete:   93/ 182  51%  skipped:   19, failed:    3ERROR   - 'qemu_x86_64/tests/kernel/early_sleep/kernel.common.sleep' generated an exception:
ERROR   - Traceback (most recent call last):
ERROR   -   File "/home/apboie/projects/zephyr/zephyr/scripts/sanity_chk/sanitylib.py", line 3072, in execute
ERROR   -     data = future.result()
ERROR   -   File "/usr/lib/python3.8/concurrent/futures/_base.py", line 432, in result
ERROR   -     return self.__get_result()
ERROR   -   File "/usr/lib/python3.8/concurrent/futures/_base.py", line 388, in __get_result
ERROR   -     raise self._exception
ERROR   -   File "/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
ERROR   -     result = self.fn(*self.args, **self.kwargs)
ERROR   -   File "/home/apboie/projects/zephyr/zephyr/scripts/sanity_chk/sanitylib.py", line 2145, in process
ERROR   -     self.cleanup_artifacts()
ERROR   -   File "/home/apboie/projects/zephyr/zephyr/scripts/sanity_chk/sanitylib.py", line 2157, in cleanup_artifacts
ERROR   -     allow += additional_keep
ERROR   - TypeError: 'NoneType' object is not iterable
INFO    - Total complete:  101/ 182  55%  skipped:   19, failed:    3
```

galak, 2020-10-20:
@andrewboie trying to reproduce this and not having success.

Are there other flags to sanitycheck hidden in `scr`?

I've hacked a working tree to create a false failure to see if its related to that, but that seems to work ok for me.

andrewboie, 2020-10-20:
`alias scr='nice sanitycheck -j36 -M -c --outdir /dev/shm/sanity-out'`

galak, 2020-10-20:
> `alias scr='nice sanitycheck -j36 -M -c --outdir /dev/shm/sanity-out'`

Cool the `-M` is where the bug is -- if you need a workaround for now, remove the `-M`.

PR #29381 (2020-10-21): sanitycheck: Fix bug with -M option
If we get a build failure with the -M option we get the following:

TypeError: 'NoneType' object is not iterable

This was due to having the default value of additional_keep in
cleanup_artifacts being None instead of an empty array.

Fixes #29376
commit eb2e89ae (2020-10-21): sanitycheck: Fix bug with -M option

If we get a build failure with the -M option we get the following:

TypeError: 'NoneType' object is not iterable

This was due to having the default value of additional_keep in
cleanup_artifacts being None instead of an empty array.

Fixes #29376
closed 2020-10-21 by galak