zephyrproject-rtos/zephyr #29394: ESP32 development overview

sylvioalves, 2020-10-21:
## Introduction

This RFC gives an overview of features and enhancements we plan to contribute for Espressif's ESP32 SoC device. It is not intended to be a proper roadmap but it gives some information about our effort on having more support for Zephyr RTOS.

### Problem description

Currently, ESP32 support consists of a few peripheral implementations as such as GPIO, UART and I2C, which limits device usage on Zephyr. There are missing peripheral drivers as such as WiFi and BLE that are very often requested by the community, but lack implementation.

### Proposed change

The proposal is to enable the powerful features that ESP32 SoC can offer, starting by bringing up WiFi subsystem. Meanwhile, we shall also work on developing yet unsupported peripheral drivers.

## Detailed RFC

### Initial Development

As there have already been some work on the WiFi bring up, a few modifications can be listed below describing initial requirements:  

* Add ESP-IDF bootloader that handles SoC initialization (cache, MMU and partition tables). In long term, Zephyr's bootloader might be supported as well.
* Set XIP support by default to support ESP-IDF bootloader.
* Update linker script with proper flash, ram sections and symbols map.
* Update clock settings and entropy drivers.
* Add timer/counter support.
* Add NVS support.
* Add Wifi sources and network drivers.

### WiFi Driver and Network Integration

We have already implemented the WiFi adapter driver, which binds ESP32 internal WiFi to Zephyr's `ETHERNET_L2` layer. However, Zephyr has yet no support for WiFI SoC that runs as `non-offloaded` driver. It means that creating a network device following `subsys/net/l2/wifi/wifi_mgmt.c` API does not allow binding to the `ETHERNET_L2`. There is no such implementation for a `WIFI_L2` layer, which would allow using `WIFI_MGMT` API to integrate with `ETHERNET_L2` layer.

This will be addressed in another issue focused on network integration and user API.

### ESP32 HAL Module

To keep the folder structure in sync with other vendors, and to update ESP-IDF source to version 4.2, repository `modules/hal/esp-idf` shall be updated to `modules/hal/espressif`. Only minimal and necessary ESP-IDF sources shall be included in this repository, which includes source files, adapter layers and symbols to handle internal calls. 
It will require changing `west.yml` to point to this new branch.

### Peripherals

The list below shows currently unsupported peripherals. We plan to start developing a few drivers in long term. Porting all the components is tricky due to FreeRTOS dependency in the current ESP-IDF implementation. SPI related code has special considerations due to dual-core SoC architecture and XIP (code execution from Flash). BLE and WiFi share the same RF radio, which implies handling their coexistence by hardware or software. ESP-IDF already takes care of everything but the porting is not straightforward. Also, we still have to analyze issues regarding multi-core support, SMP and power management.

* SPI
* SPIRAM
* BLE
* SMP
* DMA
* ULP
* ADC
* DAC
* CAN
* I2S

## Future Plans

### Future Devices

It is well knwon that Espressif has other chipsets on its plataforms. We plan to add support to them as well when a reasonable ESP32 support on Zephyr is achieved for product level usage. For the time being, there are plans to add support for (not in any chronological order):

- ESP32-S2
- ESP32-S3
- ESP32-C2
- ESP32-C3

More may be added to the list as Espressif adds more chipsets to its linecard.

ExtremeGTX, 2020-10-25:
I think after [ESP32 WiFi Support](https://github.com/zephyrproject-rtos/zephyr/issues/3723#issuecomment-698234652) a lot of basic stuff will be fixed, like WIFI, Flash Cache, 2nd Stage Bootloader support, Linker script updates.

Could you please tell us if there are any progress on the mentioned drivers ? (for cooperation and avoiding double efforts)

sylvioalves, 2020-10-26:
The items below are implemented. We are a few days of creating a PR for that:
```c
Add ESP-IDF bootloader that handles SoC initialization (cache, MMU and partition tables). In long term, Zephyr's bootloader might be supported as well.
Set XIP support by default to support ESP-IDF bootloader.
Update linker script with proper flash, ram sections and symbols map.
Update clock settings and entropy drivers.
Add timer/counter support.
Add NVS support.
Add Wifi sources and network drivers.
```

We also expect to start working on SPI for the next days.

chrta, 2020-10-26:
Could you add some information when/if Secure Boot and Flash Encryption are planned?

sylvioalves, 2020-10-28:
> Could you add some information when/if Secure Boot and Flash Encryption are planned?

Sure. I can't tell you exactly when but It is on our development backlog. I'll update you once we get ourselves working on it.

mrrosen, 2020-11-16:
Note, I added support for ESP32-S2 in a PR a while back but since it required an unreleased version of the ESP-IDF (v4.2), I stalled work on that PR until Espressif officially released a stable version supporting ESP32-S2; If youre changing the way Zephyr includes the IDF, Id be happy to adapt my PR for it but I think waiting on a stable release of it would be best
(https://github.com/zephyrproject-rtos/zephyr/pull/27246)

mahavirj, 2020-11-18:
@mrrosen 

> I added support for ESP32-S2 in a PR a while back

Thank you for update.

> If youre changing the way Zephyr includes the IDF, Id be happy to adapt my PR for it but I think waiting on a stable release of it would be best

Yes we are trying to align with stable ESP-IDF v4.2 release timelines. Release candidate is due sometimes next week, followed up by formal announcement. Regarding how Zephyr consumes IDF, some discussions are on-going at https://github.com/zephyrproject-rtos/esp-idf/issues/4, we will keep you posted on your PR for `ESP32-S2` support based on our progress.

Ithilien, 2020-12-09:
@sylvioalves Since ESP-IDF v4.2 has finally been released, what is the update plan for the esp-idf/hal_espressif so I can align my ESP32-S2 PR to it?

mrrosen, 2020-12-09:
@sylvioalves Since ESP-IDF v4.2 has been finally been released, what is the plan for updating the esp-idf/hal_espressif so I can align my ESP32-S2 PR to it?

sylvioalves, 2020-12-09:
@mrrosen, we are waiting approval to use this `hal_espressif`branch: https://github.com/sylvioalves/hal_espressif/tree/zephyr  
It is based on release/v4.2 and will probably make easier for you to rebase.
I'll update you once we get on track.

shlomow, 2020-12-12:
Hi! I saw some forks that contain some features for the esp32. Maybe it will be useful if anyone that works on features for the esp32 can post here his fork. It will remove any duplication of work and allow people to test and use some code which has not upstreamed yet. I intend to add support for spi and I hope that I am not duplicating this.

sylvioalves, 2020-12-12:
Yes, we need to handle that in some way. 
Regarding SPI, we have already implemented SPI2 and SPI3 drivers but it is under internal review. Same for timer drivers. Once we have a few PRs merged, we will push it as well.

DRNadler, 2021-01-09:
Great to hear of your progress on ESP32! In case you do not have enough to do, if I understand correctly S3 adds USB - perhaps USB support should go on the list? Thanks for your efforts!

oldrev, 2021-01-12:
Thanks for your great effort.

Is there any chance to support the SmartConfig mechanism?

K2AAE, 2021-04-23:
It's been awhile, any updates? Just purchased a ESP32-S2

tiborrr, 2021-05-25:
@sylvioalves how is the progress going on security features like secure boot and flash encryption?

sylvioalves, 2021-05-25:
> @sylvioalves how is the progress going on security features like secure boot and flash encryption?

Hi @tiborrr, our current efforts are directed to support ESP32S2 and ESP32C3. For security, so far, we haven't advanced on such front. But I invite you to attend to our lightning talk about our plans on Zephyr Summit.

tiborrr, 2021-05-26:
> Hi @tiborrr, our current efforts are directed to support ESP32S2 and ESP32C3. For security, so far, we haven't advanced on such front. But I invite you to attend to our lightning talk about our [...]

I already have registered so I will definitely join the talk. Thanks for the update!

torntrousers, 2021-06-20:
> Hi @tiborrr, our current efforts are directed to support ESP32S2 and ESP32C3. For security, so far, we haven't advanced on such front. But I invite you to attend to our lightning talk about our [...]

Is there a recording of that talk available somewhere?

tiborrr, 2021-06-20:
The recording is not yet online however the slides are. Perhaps you can find what you need in there.

https://drive.google.com/file/d/1AmP1R9Mr08nqR_wwNLsgOHaiKaLR9UlS/view?usp=sharing

torntrousers, 2021-06-20:
> 
> 
> @torntrousers The recording is not yet online however the slides are. Perhaps you can find what you need in there.
> 
> https://drive.google.com/file/d/1AmP1R9Mr08nqR_wwNLsgOHaiKaLR9UlS/view?usp=sharing

Thanks! 

I'm completly new to Zephyr and this morning have been trying to get it running on an ESP32 but really quite lost - none of the doc or guides I can find seem to work right away or be quite up to date. Is there something up to date somewhere that wold tell me how to get helloworld or blinky running on an ESP32?

tiborrr, 2021-06-20:
@torntrousers i actually switched to Ubuntu because of your experience. Getting USB to work on WSL was a pain. Now that I've switched to Ubuntu developing has been a treat.

sylvioalves, 2021-06-21:
@torntrousers Hi, after having your environment properly set, follow the guide below:    
https://docs.zephyrproject.org/latest/boards/xtensa/esp32/doc/index.html  

Let us know if you still face any issue.

torntrousers, 2021-06-21:
Thanks @sylvioalves , I have things compiling but can;t get the standard samples working, eg: https://github.com/zephyrproject-rtos/zephyr/issues/36412 and https://github.com/zephyrproject-rtos/zephyr/issues/36414

protobits, 2021-07-15:
> ### WiFi Driver and Network Integration
>
> We have already implemented the WiFi adapter driver, which binds ESP32 internal WiFi to Zephyr's `ETHERNET_L2` layer. However, Zephyr has yet no support for [...]

Hi, any progress on this? I ask since I registereded callbacks for CARRIER_UP/DOWN events to get notified when WiFi is connected/disconnected but it is problematic since sometimes the connection attempt fails and the CARRIER_DOWN event is not generated, since it was never up in the first place (I do get the DOWN event after an UP event).
I need to handle these events myself because reconnection on connection loss (via CONFIG_ESP32_WIFI_STA_RECONNECT) is not aware of a manual disconnection and attempts to reconnect again. Also, I need the events to only start tasks (such as connecting to MQTT broker) when a connection is up (otherwise I need to brute force connect attempts).

sylvioalves, 2021-08-04:
@protobits Working on it.

jli157, 2021-08-12:
@sylvioalves do you have plan to support the USB Device Mode on ESP32-S2(S3)? Thank you!

sylvioalves, 2021-08-12:
> @sylvioalves do you have plan to support the USB Device Mode on ESP32-S2(S3)? Thank you!

@jli157 Hi, thanks for asking. We have not planned that yet since we have just started ESP32S2 and ESP32C3 support. We have a long way before getting into USB device mode support, which includes low power mode, basic peripherals and so. Recently we shared our plans during Zephyr's summit. Find the link below in case you missed it.  
https://youtu.be/Lct81v9CYzg 

I'll bring more information in this channel once we update the development status.

johnnyguoyy, 2021-08-13:
Looking forward to the DMA driver!

wimmatthijs, 2021-08-28:
Could we update the overview? looking to start a new project, but not completely clear what is supported in the current version and what not....
Thanks

filipeaguiar22, 2021-09-16:
Hi, 
   I have an ESP32 board with a 26 MHz clock, so in need to configure the IDF  ESP32_XTAL_FREQ 
   I have been trying to figure out how to configure the clock, but the configurations are not in west build -t guiconfig -b esp32
   Thanks

glaubermaroto, 2021-09-16:
> Hi,
> I have an ESP32 board with a 26 MHz clock, so in need to configure the IDF ESP32_XTAL_FREQ
> I have been trying to figure out how to configure the clock, but the configurations are not in west build -t guiconfig -b esp32
> Thanks

@filipeaguiar22 It may be helpful to you knowing that the following PR was submitted: #38543

rftafas, 2021-11-03:
FWIW, updated list: https://www.espressif.com/en/news/new_operating_systems_in_ESP32

gera-k, 2021-12-27:
The cpp_synchronization sample doesn't work on esp32c3. It looks like constructors for static objects are not called.
Is it known  problem? Is the fix in the plans?

sylvioalves, 2021-12-28:
@gera-k, thanks, will take a look.

gera-k, 2021-12-28:
a question
when i do uint64_t = uint32_t * uint32_t in C, the gcc generates
mul
mulh
while RISC-V instruction manual suggests the opposite order of mul instructions when 64-bit result is required.
Is C3 CPU smart enough to do single multiplication in this case? If not, is there a way to tell gcc to generate muls in right order?

sylvioalves, 2021-12-29:
> a question when i do uint64_t = uint32_t * uint32_t in C, the gcc generates mul mulh while RISC-V instruction manual suggests the opposite order of mul instructions when 64-bit result is required. [...]

As far from a few discussion I had, it just work as is. For some reaso, when optimization is disabled, the `mul` order is indeed not following ISA recommendation. When opt. is enabled (any), it is in correct order. Anyway, it works. Have you faced any issue at all?

gera-k, 2021-12-30:
It seems it gives me wrong order with -O2.
Anyway, i tried forcing the correct order by inline assembly and it doesn't make noticeable difference in performance.

MDr164, 2022-03-28:
> Hi @tiborrr, our current efforts are directed to support ESP32S2 and ESP32C3. For security, so far, we haven't advanced on such front. But I invite you to attend to our lightning talk about our [...]

Quite a few months have passed and the support of the ESP32 in Zephyr seems to be quite ironed out. There are no Kconfigs yet, that would switch these features on itself, right? So far my path was to compile everything manually (e.g. ESP-IDF bootloader with encryption etc.) and then flash the Zephyr binary into the app partition. There is no more elegant way at the moment, or am I missing anything?

MDr164, 2022-03-28:
For unfortunate reasons we can't use MCUBoot on the current project I'm working on. It would have been the desired path but nothing I can use for right now event though it would have met all requirements.

rftafas, 2022-03-28:
> Quite a few months have passed and the support of the ESP32 in Zephyr seems to be quite ironed out. There are no Kconfigs yet, that would switch these features on itself, right? So far my path was [...]

Hi!

As stated before by @beriberikix, we already support secure boot and other boot-related features through MCUboot. Same with Flash Encryption, it is available with MCUboot. Full IDFboot support is not on our roadmap, as they are equivalent to the more 'Zephyr standard' MCUboot features.

As @sylvioalves said, but adding that it is due to such equivalence, our efforts are now on PinCtrl, Low Power, SOC (S2/S3) support and AMP (AMP is currently part of our S3 support efforts).

That said, as for IDFboot: it's possible to use it with some tinkering, though, for those with an imperative need for both IDFboot and Zephyr. We haven't prevented that in any form. Further for my understanding: the only scenario that I can imagine for such situation (mandatory IDFboot) is a product that was shipped with ESP-IDF image and must move to Zephyr. I would be glad if you could share your scenario.

maximevince, 2022-04-07:
Is there visibility on the Low Power support for ESP32 on Zephyr? (and especially interested in the C3)
Is this started / planned / available for testing / ... ?

beriberikix, 2022-04-11:
Would it be possible to create an up to date table of the supported features like what was shared on the [Espressif Blog](https://www.espressif.com/en/news/new_operating_systems_in_ESP32)? Thank you!

wolterhv, 2022-04-15:
Hi, I'm interested in adding support for the ESP32 ADC. I've opened #44908 for this purpose. Any willingness to collaborate is very welcome.

sam131208, 2022-04-19:
@sylvioalves Do you have any plans to release S3 support?

henrikbrixandersen, 2022-04-21:
I have started work on adding support for the ESP32 CAN controller.

rftafas, 2022-04-22:
> @sylvioalves Do you have any plans to release S3 support?

ICYMI:
https://github.com/zephyrproject-rtos/hal_espressif/issues/97

maximevince, 2022-04-22:
> > Is there visibility on the Low Power support for ESP32 on Zephyr? (and especially interested in the C3) Is this started / planned / available for testing / ... ?
> 
> We completed one milestone related to this (PinControl). Now we are planning and doing the priority check against other tasks. Any specifics you would like to share? Need for deep sleep? Do you intend to control the ULP / write code for it? Since ULP is different on ESP32 and ESP32S2, any target?

Especially getting the device into deep sleep from a Zephyr application is the first major goal to getting to a viable battery powered device (talking multiple years on a primary battery cell). Thanks for the replies here, much appreciated!!

GuptaAshi-Eaton, 2022-05-10:
Hello,

By when we can expect support for ESP32-S3 in Zephyr ? As per timeline given on https://www.espressif.com/en/news/new_operating_systems_in_ESP32 this would have available by Q1. Is there any new timeline ?

sam131208, 2022-05-19:
@rftafas 

You mentioned the low power consumption of the ESP32, which is very important in product deployment. When can we test it?

mickeyl, 2022-06-01:
> I have started work on adding support for the ESP32 CAN controller.

Awesome, is there any progress yet?

rftafas, 2022-06-01:
FWIW: https://www.espressif.com/en/news/Zephyr_updates

Red-M, 2022-06-02:
I am interested to see the S3 supported for some various reasons of my own. Mainly looking for a dual core MCU with 34+ GPIO pins and USB device support.

enthunilu, 2022-07-04:
Hello,

Is there a near future plan to have USB support for ESP32-S2/S3 ?

Suhas-MS, 2022-08-02:
Hello,
I have a requirement for SPI slave using ESP32. Is this supported now or will it added in the future?

rftafas, 2022-08-02:
> Hello, I have a requirement for SPI slave using ESP32. Is this supported now or will it added in the future?

@Suhas-MS, it's not supported yet, it's on backlog, it's not on any schedule/roadmap yet. I suggest creating a request for that or DIY if you are on a tight schedule. Usual advice to rely on already released features still apply.

sam131208, 2022-09-24:
Hello,
    Are there any plans to support I2S? I wrote a simple I2S driver before, but when I ported from zephyr ver3.1 to ver3.2, it didn't work correctly. I hope zephyr can add I2S support to the list.

rftafas, 2022-09-26:
> Hello, Are there any plans to support I2S? I wrote a simple I2S driver before, but when I ported from zephyr ver3.1 to ver3.2, it didn't work correctly. I hope zephyr can add I2S support to the list.

@sam131208 , it is not on the horizon yet and I wouldn't hold my breath either. I strongly encourage you to share your work on GH. It may be a starting point to fix your issue, points you towards a nice contribution and you may have it working earlier...

sslupsky, 2022-12-08:
Is there support for ESP32-C6?

rftafas, 2022-12-08:
Not yet. Unless you need specifically for 802.15.4 and Wi-Fi 6, you can do fine with C3.

rftafas, 2023-01-11:
ICYMI: https://github.com/zephyrproject-rtos/zephyr/pull/53699

[113 more replies]

closed 2025-11-04 by sylvioalves