zephyrproject-rtos/zephyr #29397: Build all tests of module mcuboot

nvlsianpu, 2020-10-21:
**only test for frdm_k64f board are build for mcuboot project**
I expected that all test for which `platform_allow:`  in https://github.com/zephyrproject-rtos/mcuboot/blob/master/boot/zephyr/sample.yaml of should be tested.
For MCUBoot nrf52840 boards are used as target for integration test (all of them are expected to be build-only)

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

PerMac, 2020-11-26:
With the #29829 merged this should be possible by adding platforms that need to be tested (build level in case of not simulated ones) in 'integration_platforms` list. This list expands/limits the CI to the platforms in the list (`platform_allow` only works as an additional filter preventing anything that is not on the list from being tested).

nvlsianpu, 2021-05-28:
@PerMac Have to check again, but as i remember mcuboot was not build even with these hints. Maybe because it is external application.

nordic-krch, 2021-05-28:
Just wanted to mention that was just fixing an issue with mcuboot failing to compile in certain configuration (reported by downstream project) #35765. It is not verified by zephyr CI though.

nvlsianpu, 2021-06-01:
Can building of MCUboot can be added to the CI? I think additional twister run with `--testcase-root` set to `bootloader/mcuboot/boot/zephyr` will do the job.

@carlescufi Can this be considered and implemented? Maybe this need TSC approval.

Here draft with modified sample.yaml
https://github.com/mcu-tools/mcuboot/pull/890/

PR #30285 (2021-06-11): manifest: mcuboot: allow to build nRF52840 targets in CI
MCUboot was build only for frdm_k64f in zephyr CI.
Extended Zephyr-RTOS integration platform by nrf52840dk_nrf52840
and nrf52840dongle_nrf52840 which allow to build MCUBoot on
these platform in Zephyr-RTOS CI.

https://github.com/mcu-tools/mcuboot/pull/890

Also other improvements within the mcuboot synchronization:

* Added AES256 support for image encryption
* zephyr: serial: Remove unnecessary call to irq_unlock
* boot_serial: Port encoding to use cddl-gen
* bootutil_public: Make boot_read_swap_state declaration public

fixes #29397
closed 2021-06-11 by carlescufi