zephyrproject-rtos/zephyr #29414: west debug under WSL2

oxymoron87, 2020-10-21:
Hi,
I am trying to use `west debug` under WSL2 with `jlink` runner (JLink is installed under Windows because only under Windows JLink has access to the USB ports where debug probes are).
For this I had to make a symlink 
/usr/local/bin/JLinkGDBServer -> '/mnt/c/Program Files (x86)/SEGGER/JLink/JLinkGDBServer.exe'
(Adding /mnt/c/Program Files (x86)/SEGGER/JLink/ to the $PATH didn't solve the problem because west tried to run JLinkGDBServer not the JLinkGDBServer.exe.)

After that I've realized that `--tool-opt="-noLocalhostOnly"` is needed for GDB server to listen to all interfaces (not only localhost) because it actually runs under Windows (localhost in Windows is different from localhost in WSL).

But still I ended up with 
> :2331: Connection timed out

To solve this I had to modify zephyr/scripts/west_commands/runners/jlink.py and change `target remote` command to include  Windows instance's WSL interface IP address
`client_cmd = (self.gdb_cmd +
                          self.tui_arg +
                          [self.elf_name] +
                          ['-ex', 'target remote <Windows instance's WSL interface IP address>:{}'.format(self.gdb_port)])`

Suggestion:
It would be great if west is able to identify that it runs under WSL and run GDB server automatically with option `-noLocalhostOnly` and GDB client with IP address of the Windows instance (this can be retrieved from /etc/resolv.conf file from nameserver variable). 
Or at least have an option to specify IP address which gdb client should use, e.g. `--gdb-ip` (like we have `--gdb-port` now).

It could also solve the issue zephyrproject-rtos/west#228 by using debug directly.

mbolivar-nordic, 2020-10-21:
Transferred to a zpehyr issue since west debug is an extension defined in the zephyr repo.

In general we don't support WSL2 and would need to test various workflows to claim support for it, so I'm marking this as an enhancement.

mmurty, 2021-09-18:
Hi @mbolivar ,
I am trying to do the same thing in WSL.
Any update on this issue?

matt-wood-ct, 2024-05-11:
IDK if you guys know this, but you can pass the whole USB device down into WSL as if it was natively connected to the linux environment, see this guide/doc from Microsoft: https://learn.microsoft.com/en-us/windows/wsl/connect-usb