zephyrproject-rtos/zephyr #29418: ieee802154: cc13xx_cc26xx: bug in rf driver library

cfriedt, 2020-10-21:
**Describe the bug**
Filing this here because [hal/ti](https://github.com/zephyrproject-rtos/hal_ti) does not have an issue tracker and because it affects #26316.

In the process of switching to use the RF Driver API from the SimpleLink SDK (Issue: #26312 PR #26316), @vanti & I observed a scenario where packet transmission would suddenly stop.

I traced that to an issue related to some of the internal workings of TI's RF Driver API and brought the bug to TI's attention internally. It is **very** loosely related to modulo arithmetic, which makes the error observable within a predictable time period.

The issue is 100% reproducible.

It is **not** an issue that has any immediate implications about device security, and only affects functionality, just to clarify.

**To Reproduce**
1. On one launchpad (or really any other supported IEEE 802.15.4 board), flash the `echo_server` example from `master`.
2. On another launchpad, flash the `echo_client` example from branch in the PR which uses TI's RF Driver API.
3. Observe the console of the `echo_client`.
4. Within a few minutes, it becomes apparent that messages are not transmitted, because the following is repeated:
```
<err> net_echo_client_sample: UDP IPv6: Data packet not received
```
5. On the console of the `echo_client` board, issue the command `ieee802154 set_chan 11`, followed by the command `ieee802154 set_chan 26`. This effectively clears the internal radio command queue.
6. Observe that TX and RX continue to work again properly, until the 

**Expected behavior**
UDP packets should continue being sent and received indefinitely.

**Impact**
It's a showstopper.

**Logs and console output**
[bug.txt](https://github.com/zephyrproject-rtos/zephyr/files/5417576/bug.txt)

**Environment (please complete the following information):**
 - OS: Zephyr
 - Toolchain: Zephyr 0.11.3 SDK
 - Commit SHA: cfb6257327c763b6540475da3f0a8f83b67e5b3e

**Additional context**
While it is possible to wait for a new release of the SimpleLink SDK, the proposed solution involves a temporary workaround, as several other issues with PR's ready would be blocked for an indefinite amount of time (e.g. #26315, #21631).

The workaround does programmatically what was done manually - it flushes the radio command queue close to when command ID numbers are to wrap-around the 12-bit modulus (4096).

The workaround has already shown to be effective (see attached) and the `echo_server` / `echo_client` application again sends and receives messages indefinitely.
[ti.txt](https://github.com/zephyrproject-rtos/zephyr/files/5418021/ti.txt)

SML892, 2020-10-23:
Hi @cfriedt  thanks for the analysis.

As you have observed, the sequence numbers will wrap at 12-bit max.
This because they are incremented using the code below (from RFCC26X2_multiMode.c):

```c
/* Increment the sequence number and mask the value */
RF_cmdQ.nSeqPost = (RF_cmdQ.nSeqPost + 1) & N_CMD_MODMASK;
```

Where `N_CMD_MODMASK` is 0xFFF.
As you have observed, if there are no errors, the command handle will be assigned to that sequence number.
However, the `RF_cmdHandle` is not guaranteed to be linearly increasing. From `RFCC26X2.h`, it is only stated that they will be unique and greater than 0 in the case of no errors.

I have filed a ticket to increase the range of sequence numbers (and thus command handles) before wraparound, however I would warn against relying on the command handle as a sequence number.

Again, thank you very much for bringing this issue to our attention. We'll follow up on the JIRA internally.

cfriedt, 2020-10-23:
@SML892 - simply wrapping around (at 0) is not the actual bug in this case and, yes, that is (some of) the driver code performing modulo arithmetic.

It is slightly more nuanced than that. Increasing the modulus would not solve the actual problem. I am already working with @vanti and others at TI to solve the bug in the SimpleLink SDK).

The information is provided here simply  to demonstrate that there is a work-around and to un-block several other pieces of radio related work that depend on using TI's RF driver API (adding support for Sub GHz, RAW mode for 2.4 GHz, RAW mode for Sub GHz, simultaneous BLE support, ..).

SML892, 2020-10-23:
@cfriedt - thanks for information.

> The actual bug is slightly more nuanced than that. Increasing the modulus would not solve the actual problem.

Can you explain further? From your description above it seems that the issue is an unexpected wraparound of the command handle. If this is not actually the issue you're seeing, please let me know. Furthermore, if you have any more information regarding that can point to the root cause of the issue please provide.

> I am already working with @vanti and others at TI to solve the bug in the SimpleLink SDK

My team owns and maintains the RF driver that is shipped as part of the cc13x2_26x2 SDK and others.
Any information you have will be helpful for us to correctly file the issue and create some regression tests.

cfriedt, 2020-10-23:
> @cfriedt - thanks for information.
>
>
> Can you explain further? From your description above it seems that the issue is an unexpected wraparound of the command handle. If this is not actually the issue [...]

It is expected that the command handle wraps around but there is a corner case that is currently not handled. I'm afraid that's all I can disclose at the moment. 

> > I am already working with @vanti and others at TI to solve the bug in the SimpleLink SDK
> 
> My team owns and maintains the RF driver that is shipped as part of the cc13x2_26x2 SDK and others.
> Any information you have will be helpful for us to correctly file the issue and create some regression tests.

Maybe I've emailed the wrong manager then (Shmulik?).

> Edit: upon further review, it seems that the real issue is that there is a "scenario where packet transmission would suddenly stop".
> To help us further characterize this can you answer the following:

Fixing the SDK bug is outside the scope of the my current contract.

I do already have a solution already and have offered my professional services to TI to help to resolve it. Also I will check shortly to see if the bug is still present in v4.30. 

I also emailed the existing manager from TI who is on the Zephyr board as well as the current cc13xx_cc26xx SW manager about this. I asked if TI would be willing to extend a fixed-cost contract to evaluate the fix that I have for the SDK and to also continue writing the BLE split-stack driver for Zephyr and am waiting to hear a reply.

BLE split-stack support in Zephyr would be a great addition to Zephyr and would certainly increase sales, so hopefully I can get the go ahead to do that.

SML892, 2020-10-23:
@cfriedt - Thanks let's follow up offline.

Should you have any more information regarding the issue itself, please feel free to post it here should anyone arrive by way of google.

PR #26316 (2020-11-03): drivers: ieee802154: cc13xx_cc26xx: adopt hal/ti rf driverlib
This change reworks the cc13xx_cc26xx IEEE 802.15.4 driver to use the TI RF driver API that is available in modules/hal/ti.

There are a number of benefits to using TI's API including
 - a stable multi-OS vendor library and API
 - API compatibility with the rest of the SimpleLink SDK and SoC family
 - potential multi-protocol & multi-client radio operation (e.g. both 15.4 and BLE)
 - coexistence support with other chipsets via gpio
 - vetted TI RF driver resources, such as
   - the radio command queue
   - highly tuned / coupled RTC [... cut]
commit bb7c58f6 (2020-11-03): ieee802154: cc13xx_cc26xx: workaround for issue in ti driver

This is a temporary workaround for an issue in TI's RF Driver
API. A subsequent release of the SimpleLink SDK will mitigate
the need for it and it can be reverted when hal/ti receives
that update.

Fixes #29418
closed 2020-11-03 by carlescufi