zephyrproject-rtos/zephyr #29435: SDCard via SD/SDIO/MMC interfaces

haipas, 2020-10-22:
**Is your feature request related to a problem? Please describe.**
Currently, an SD card is accessed via the SPI interface. Many controllers have a native SD/SDIO/MMC interface (STM32F4, STM32H7 and many more). This allows up to 4 times higher data rates when reading and writing to the SD card.

**Describe the solution you'd like**
Access to the SD card via the native interface of the controller and transfer the data from RAM via DMA to SD card.

erwango, 2020-10-22:
@haipas There's already a `subsys/disk/disk_access_stm32_sdmmc.c` driver. Can you check if this could match your request ?

erwango, 2020-10-23:
@haipas Please let us know if this answered your point and if the point could be closed.

haipas, 2020-10-23:
@erwango thanks for the hint. #29480 is a PR which adds the sdmmc support for the nucleo h743zi

closed 2020-11-09 by haipas