zephyrproject-rtos/zephyr #29438: Telnet shell do not work with NET_SOCKET_OFFLOAD

5frank, 2020-10-22:
**Is your enhancement proposal related to a problem? Please describe.**
Telnet shell do not work with offloaded sockets. i.e. not possible to have telnet shell with nrf91 modem (and I believe ublox sara-R4 modem).

**Describe the solution you'd like**
Rewrite shell_telnet to use sockets instead of net_context interface.

**Describe alternatives you've considered**
Add functionality so that net_context interface work with offloaded sockets. Might be problematic as only parts of the interface can be implemented?

**Additional context**
Related issue: https://github.com/zephyrproject-rtos/zephyr/issues/25858

I noted that `net_context_bind` etc do not give a error if there is no backend. Not sure it qualifies as a bug?

jukkar, 2020-10-22:
> I noted that `net_context_bind` etc do not give a error if there is no backend. Not sure it qualifies as a bug?

Could you elaborate this a bit more, not sure what error you mean here?
We do have asserts in `include/net/net_offload.h` for various parameters. Actually those asserts might be better to change to normal error checks instead (perhaps using the `CHECKIF()` macro).

5frank, 2020-10-23:
@jukkar with nrf91 modem sockets. I had the following:
```
NET_SOCKET_OFFLOAD=y
CONFIG_NET_NATIVE=y
CONFIG_NET_OFFLOAD=y
```
 `__net_if_dev_nrf91_socket_0.offload` is NULL and `net_if_is_socket_offloaded()` never checked.  `net_context_bind()` returned 0 with debug message:
```
  net_ctx.net_context_bind: (0x2002473c): Context 0x200231c8 binding to TCP 0.0.0.0:22 iface 0x20021fa0
[where 0x20021fa0 is __net_if_nrf91_socket_0]
```
`nrf91_socket_offload_bind()` never called, which I now learned is expected behavior, but it took me a some time to find the issue as `net_context_bind()` never complained about a missing backend/driver.

I believe it is the same behavior for `net_context_accept()`.

me-axentia, 2021-11-16:
This is an issue for us. We want to have a telnet shell on the nRF91 (using offloading sockets).
We will probably write our own telnet backend to get it working. Is it desired convert the official telnet backend to use BSD sockets instead? If so maybe we could make a PR on this when we have our solution up and running.

jukkar, 2021-11-16:
> Is it desired convert the official telnet backend to use BSD sockets instead? If so maybe we could make a PR on this when we have our solution up and running.

:+1: It would be great if the telnet backend would support BSD sockets, so if you do it, please create a PR for it.

rlubos, 2024-03-14:
Forgot to link the PR, resolved with https://github.com/zephyrproject-rtos/zephyr/pull/69612.