zephyrproject-rtos/zephyr #29440: Missing hw-flow-control; in hci_uart overlay files

koffes, 2020-10-22:
**Describe the bug**
I attempted to get HW flow control working on the nRF5340.
`hw-flow-control` is missing in the overlay of some boards in https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/bluetooth/hci_uart/boards

1. https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/bluetooth/hci_uart/boards/nrf5340pdk_nrf5340_cpuapp.overlay has this enabled, but on the APP core instead of NET.
2. The nRF5340 DK should be added in addition to the PDK

As a general question. Why is not `hw-flow-control` enabled by default on more boards and simply deactivated using CONFIG_UART_LINE_CTRL=n ?

**Impact**
UART with flow control cannot be used for hci_uart on some boards.

nordic-krch, 2020-10-22:
so `nrf5340pdk_nrf5340_cpuapp` configuration does not make sense. It should be `nrf5340pdk_nrf5340_cpunet`.

nordic-krch, 2020-10-22:
fyi @carlescufi

carlescufi, 2020-10-26:
> so `nrf5340pdk_nrf5340_cpuapp` configuration does not make sense. It should be `nrf5340pdk_nrf5340_cpunet`.

Yes it does make sense, unless I'm mistaken. The net CPU runs the controller but the main app CPU runs the hci_uart `main.c`, right @joerchan ?

carlescufi, 2020-10-26:
Agree that we are missing the overlay for the DK (we only have it for the PDK).

joerchan, 2020-10-26:
Haven't tested this combination yet. Is this with hci-uart on the application core and then hci-rpmsg on network core then?

koffes, 2020-10-26:
This particular issue was with APP core idle and hci-uart on NET. I.e. the host running on a PC.

koffes, 2021-01-05:
Stale label should be removed.

carlescufi, 2021-01-14:
@anangl and @nordic-krch could you please weigh in here?

carlescufi, 2021-02-03:
> This particular issue was with APP core idle and hci-uart on NET. I.e. the host running on a PC.

OK, then we will fix this particular case.

PR #31932 (2021-02-03): Bluetooth: samples: hci_uart: Add overlays for nRF5340
Add overlays for the current nRF5340 DK boards:

- nrf5340dk_nrf5340_cpuapp
- nrf5340dk_nrf5340_cpunet

Fixes #29440.
commit 1454730f (2021-02-03): Bluetooth: samples: hci_uart: Add overlays for nRF5340

Add overlays for the current nRF5340 DK boards:

- nrf5340dk_nrf5340_cpuapp
- nrf5340dk_nrf5340_cpunet

Fixes #29440.
closed 2021-02-03 by nashif