zephyrproject-rtos/zephyr #29442: build failure w/sanitycheck for samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4

galak, 2020-10-22:
On multiple platform seeing failures if we do:

`./scripts/sanitycheck -s samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 -l`

we get:
```
ERROR   - bl654_dvk                 samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/bl654_dvk/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log
INFO    - Total complete:    1/  16   6%  skipped:    0, failed:    1
ERROR   - nrf52840dk_nrf52840       samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/nrf52840dk_nrf52840/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log
INFO    - Total complete:    2/  16  12%  skipped:    0, failed:    2
ERROR   - bl653_dvk                 samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/bl653_dvk/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log
INFO    - Total complete:    3/  16  18%  skipped:    0, failed:    3
ERROR   - particle_boron            samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/particle_boron/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log
INFO    - Total complete:    4/  16  25%  skipped:    0, failed:    4
ERROR   - reel_board_v2             samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/reel_board_v2/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log
INFO    - Total complete:    5/  16  31%  skipped:    0, failed:    5
ERROR   - nrf52833dk_nrf52833       samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/nrf52833dk_nrf52833/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log
INFO    - Total complete:    6/  16  37%  skipped:    0, failed:    6
ERROR   - pinnacle_100_dvk          samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/pinnacle_100_dvk/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log
INFO    - Total complete:    7/  16  43%  skipped:    0, failed:    7
ERROR   - nrf52840dongle_nrf52840   samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/nrf52840dongle_nrf52840/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log
INFO    - Total complete:    8/  16  50%  skipped:    0, failed:    8
ERROR   - particle_xenon            samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/particle_xenon/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log
INFO    - Total complete:    9/  16  56%  skipped:    0, failed:    9
ERROR   - nrf52840_blip             samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/nrf52840_blip/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log
INFO    - Total complete:   10/  16  62%  skipped:    0, failed:   10
ERROR   - reel_board                samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/reel_board/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log
INFO    - Total complete:   11/  16  68%  skipped:    0, failed:   11
ERROR   - particle_argon            samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/particle_argon/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log
INFO    - Total complete:   12/  16  75%  skipped:    0, failed:   12
ERROR   - nrf52840_papyr            samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/nrf52840_papyr/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log
INFO    - Total complete:   13/  16  81%  skipped:    0, failed:   13
ERROR   - nrf52840_mdk              samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/nrf52840_mdk/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log
INFO    - Total complete:   14/  16  87%  skipped:    0, failed:   14
ERROR   - adafruit_feather_nrf52840 samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/adafruit_feather_nrf52840/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log
INFO    - Total complete:   15/  16  93%  skipped:    0, failed:   15
ERROR   - nrf52833dk_nrf52820       samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/nrf52833dk_nrf52820/samples/bluetooth/hci_usb_h4/sample.bluetooth.hci_usb_h4/build.log

```

From the log:
```
/home/galak/git/zephyr/subsys/bluetooth/controller/ll_sw/ull_scan.c: In function 'll_scan_enable':
/home/galak/git/zephyr/subsys/bluetooth/controller/ll_sw/ull_scan.c:130:10: error: unused variable 'is_update_1m' [-Werror=unused-variable]
  130 |  uint8_t is_update_1m = 0U;
      |          ^~~~~~~~~~~~
/home/galak/git/zephyr/subsys/bluetooth/controller/ll_sw/ull_scan.c:127:10: error: unused variable 'is_update_coded' [-Werror=unused-variable]
  127 |  uint8_t is_update_coded = 0U;
      |          ^~~~~~~~~~~~~~~
/home/galak/git/zephyr/subsys/bluetooth/controller/ll_sw/ull_scan.c:125:22: error: unused variable 'node_rx_scan_term' [-Werror=unused-variable]
  125 |  struct node_rx_pdu *node_rx_scan_term = NULL;
      |                      ^~~~~~~~~~~~~~~~~

```

galak, 2020-10-22:
(also see same failures with `samples/bluetooth/hci_usb/sample.bluetooth.hci_usb` and a number of tests for bbc_microbit.

cvinayak, 2020-10-23:
@galak tests/bluetooth/init  with CONF_FILE=prj_ctlr_4_0.conf should be run on normal PR CI, but looks like the offending PR did not fail CI. The tests is not built in https://github.com/zephyrproject-rtos/zephyr/pull/27365 ? How to ensure it is built in the future so that new feature developments that introduce compile failures are caught when submitting PRs?

galak, 2020-10-23:
> @galak tests/bluetooth/init with CONF_FILE=prj_ctlr_4_0.conf should be run on normal PR CI, but looks like the offending PR did not fail CI. The tests is not built in #27365 ? How to ensure it is [...]

Adding @nashif to that query as he's more familiar with how the sanitycheck scripts determine what tests to run.

PR #29462 (2020-10-26): Bluetooth: controller: Fix arguments unused without Extended Scanning
Fix arguments unused when not enabling Extended Scanning which
was introduced in commit 0cef1e43c9

Fixes #29442.
commit c74438cd (2020-10-26): Bluetooth: controller: Fix arguments unused without Extended Scanning

Fix arguments unused when not enabling Extended Scanning
which was introduced in
commit 0cef1e43c915 ("Bluetooth: controller: Extended
Scanning Coded PHY duration and period").

Fixes #29442.
closed 2020-10-26 by carlescufi