zephyrproject-rtos/zephyr #29444: Network deadlock

ohitz, 2020-10-22:
**Describe the bug**

There is another network deadlock with TCP. It is possible that `net_tcp_put()` waits for the TCP mutex forever and locks the entire network stack. The source of the problem is probably the same as in #29347, locking two mutexes not always in the same order.

**To Reproduce**

The deadlock doesn't occur very often, but it can be forced by introducing a short delay into `net_tcp_put()` right before locking the mutex. With that modification, the `samples/net/sockets/echo_server` example program can be used to reproduce it:

1. Patch `net_tcp_put()` as follows:

```
diff --git a/subsys/net/ip/tcp2.c b/subsys/net/ip/tcp2.c
index 9605d26eed..1fcc04bf0e 100644
--- a/subsys/net/ip/tcp2.c
+++ b/subsys/net/ip/tcp2.c
@@ -1523,6 +1523,7 @@ int net_tcp_put(struct net_context *context)
                return -ENOENT;
        }
 
+       k_sleep(K_MSEC(10));
        k_mutex_lock(&conn->lock, K_FOREVER);
 
        NET_DBG("%s", conn ? log_strdup(tcp_conn_state(conn, NULL)) : "");
```

2. Compile the echo server and flash it
3. Open a connection to the echo server and close it immediately: `echo "" | nc -N 192.0.2.1 4242`

After this, the complete network stack is locked.

**Environment (please complete the following information):**

 - Zephyr f6042bbd34d310745604f4488ad416b03f94c946
 - MIMXRT1060-EVK eval board

jukkar, 2020-10-22:
Thanks Oliver, I will investigate this one.

armandciejak, 2020-10-23:
I think there is a design issue here. As explained in #29347 there are 2 mutexes used in the TCP logic, one for `struct tcp` and one for the `struct net_context` but these 2 structures are bound together. I my opinion these 2 mutex are mostly redundant. In the sending path and receiving path they are naturally locked in reverse order which leads to the deadlocks that @ohitz reported.

I think that in order to definitively get rid of the deadlocks, 2 mutexes should not be used because this is the root cause of the problem. One could either use a single mutex (if possible) or use other synchronization mechanisms.

@jukkar You know this code much better than us, what do you think?

jukkar, 2020-10-23:
> You know this code much better than us, what do you think?

Perhaps that could be done, I am not fond of having two locks here either. The net_context mutex is definitely needed, but perhaps the tcp one could be omitted. This needs more investigation to be certain.

Anyway, I think I managed to fix this one. When closing the connection, we had tcp locks held which caused this mixed locking ordering.

ohitz, 2020-10-23:
Thanks for your work Jukka!

As far as I can see, that deadlock is now solved. But when I sent heavy TCP and ICMP traffic to our application, it hung again after a couple of minutes. Not sure it is a network/TCP deadlock again, but it at least looks this way. I'll investigate and open new issues when I can find reproducible deadlocks.

Maybe it would really be better to check if it would be possible to get rid of the TCP mutex altogether as Armand wrote.

jukkar, 2020-10-23:
> Maybe it would really be better to check if it would be possible to get rid of the TCP mutex altogether as Armand wrote.

Indeed, I will investigate this next.

jukkar, 2020-10-23:
@ohitz I made a quick experiment in #29481 where I removed separate lock in TCP2 and converted it to use the same lock as in net_context. Seems to work ok in quick testing, will try with heavy traffic next.

ohitz, 2020-10-23:
> @ohitz I made a quick experiment in #29481 where I removed separate lock in TCP2 and converted it to use the same lock as in net_context. Seems to work ok in quick testing, will try with heavy traffic next.

Great, that sounds really promising! I'll give it a try as well.

PR #29469 (2020-11-04): net: tcp2: Avoid deadlock when closing the connection
When a connection is being closed, it is possible that the application
will have a lock to net_context and TCP2 connection lock. If we then
receive a final TCP2 ACK and close the connection, the locking order
get switched and TCP2 will first try to get its own lock and then the
net_context lock. This will lead to deadlock as the locking ordering
is now mixed.

The solution is to unref the TCP connection after releasing the
connection lock. The TCP connection unref function will anyway get the
lock so no need to do double locking.

Fixes #29 [... cut]
commit f50c9e11 (2020-11-04): net: tcp2: Avoid deadlock when closing the connection

When a connection is being closed, it is possible that the application
will have a lock to net_context and TCP2 connection lock. If we then
receive a final TCP2 ACK and close the connection, the locking order
get switched and TCP2 will first try to get its own lock and then the
net_context lock. This will lead to deadlock as the locking ordering
is now mixed.

The solution is to unref the TCP connection after releasing the
connection lock. The TCP connection unref function will anyway get the
lock so no need to do double locking.

Fixes #2 [... cut]
closed 2020-11-04 by jukkar