zephyrproject-rtos/zephyr #29446: pwm: stm32: output signal delayed

simonguinot, 2020-10-22:
**Describe the bug**
I encountered this bug by using PWM (timer 3, channel 3) on a STM32F070 MCU. With the default prescaler configuration (10000), and when the PWM is configured to output a signal with a rather big period (around 100ms for example), then sometimes nothing happens... But after a delay (around 10 seconds), then the PWM signal is finally outputted correctly

**To Reproduce**

The following code should allow to reproduce the issue:
```
pwm_pin_set_usec(dev, pwm, 1000000, 500000, flags);
k_sleep(K_MSEC(5000));
pwm_pin_set_usec(dev, pwm, 100000, 50000, flags);
```

You should observe that the first PWM configuration is applied correctly. But the second one is delayed (of around 10 seconds). During that time no PWM signal is outputted on pin. The previous PWM period/pulse is not active anymore. And after the delay, the second PWM signal is finally outputted correctly.

**Expected behavior**
The PWM configuration should be applied without any delay.

**Impact**
Among other things this bug affects the driving of PWM LEDs. Sometimes the LED blinking effects are delayed. 

**Additional context**
It seems that decreasing the prescaler works around the issue. With a prescaler value set to 1000, I can't reproduce it anymore.

bpbradley, 2020-10-22:
Just chiming in to mention I have experienced this exact same issue on stm32f4 with the default prescaler. I can try to reproduce with your method soon.

gmarull, 2020-10-22:
@simonguinot what is the MCU clock speed?

simonguinot, 2020-10-22:
> @simonguinot what is the MCU clock speed?

It is 48MHz. So with the default prescaler (10000) the PWM is working at ~4800 cycles per second. On STM32F070, the timer 3 is a 16-bit counter. Then the maximum period is between 13 and 14 seconds.

gmarull, 2020-10-22:
@simonguinot can you try without preload? https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/pwm/pwm_stm32.c#L231

simonguinot, 2020-10-22:
> @simonguinot can you try without preload? https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/pwm/pwm_stm32.c#L231

I observe the same issue with or without preload.

simonguinot, 2020-10-22:
BTW, I am rusing the led-pwm sample from #29401 to reproduce the issue. At some point the LED is behaving incorrectly, staying off for long periods.

gmarull, 2020-10-22:
@simonguinot I've tried the following code on a NUCLEO-H743ZI board

```c
while (1) {
	pwm_pin_set_usec(pwm, ch, 1000000, 500000, 0);
	k_sleep(K_MSEC(5000));
	pwm_pin_set_usec(pwm, ch, 100000, 50000, 0);
	k_sleep(K_MSEC(5000));
}
```

And this is roughly what I get:

[image]

I see no big gaps where the PWM signal is low.

simonguinot, 2020-10-22:
It looks good @gmarull.

Please can you have another try with this:

```
while (1) {
        pwm_pin_set_cycles(pwm, ch, 100, 0);
        k_sleep(K_MSEC(1000));
        pwm_pin_set_cycles(pwm, ch, 100, 100);                 
        k_sleep(K_MSEC(1000));
	pwm_pin_set_usec(pwm, ch, 1000000, 500000, 0);
	k_sleep(K_MSEC(5000));
	pwm_pin_set_usec(pwm, ch, 100000, 50000, 0);
	k_sleep(K_MSEC(5000));
}
```

For me, this takes 3-4 loops to trigger the issue.

PR #29619 (2020-10-30): drivers: pwm: pwm_stm32.c: enable ARR preload
drivers: pwm: pwm_stm32.c: enable ARR preload

Enable ARR preload so that period or pulse updates are taken
into account synchronously with update event
(at the end of a ongoing period)
And thus avoid undetermined intermediate pulse.

Fixes #29446

Tested on Nucleo_f401re
closed 2020-10-30 by carlescufi