zephyrproject-rtos/zephyr #29476: Samples: TF-M: Add PSA FF API Sample

microbuilder, 2020-10-23:
In order to certify Zephyr + TF-M + a specific board for PSA FF API compliance, the results of the PSA FF API tests must be send to the certifying lab: [ARM-software/psa-arch-tests](https://github.com/ARM-software/psa-arch-tests).

These can be manually run, but adding a sample application that runs these tests on the desired board target as a standard Zephyr+TF-M build will facilitate future testing.

ioannisg, 2020-10-26:
FYI @oyvindronningstad

ioannisg, 2021-04-23:
Isn't this done now @oyvindronningstad , @microbuilder ?

oyvindronningstad, 2021-04-23:
My PR added support for the dev APIs test. FF tests is still not available.

adrianlshaw, 2021-04-23:
Just mentioning here that the Dev APIs are all you need for Functional API certification. See here https://www.psacertified.org/getting-certified/functional-api-certification/ The FF ones are good to have but not strictly needed.

microbuilder, 2021-04-23:
> Just mentioning here that the Dev APIs are all you need for Functional API certification. See here https://www.psacertified.org/getting-certified/functional-api-certification/ The FF ones are good to have but not strictly needed.

Thanks for the clarification Adrian.

microbuilder, 2021-04-23:
@ioannisg @oyvindronningstad I'm OK for now with only having the regression tests, and not the individual test suites, since facilitating PSA certification was my main goal. The extra tests are nice to have, but I'd consider them bonus points. Feel free to close this if you'd like to get it off the ToDo list.

closed 2021-05-03 by microbuilder