zephyrproject-rtos/zephyr #29495: SD card slow write SPI/Fatfs/stm32

alijafarideveloper, 2020-10-23:
**Is your enhancement proposal related to a problem? Please describe.**
A clear and concise description of what the problem is.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

alijafarideveloper, 2020-10-23:
I want to write 1MB of data on SD card.
My data comes from ADC and I must write it directly to the SD card.
Main problem is I can't write data in Realtime operation.
Max speed I need is 2MB/s but I can get just about 3KB/s. 
My hardware : 1) stm32f103c8  2)SPI SD card module

EminYagmahan, 2020-10-26:
An SD card via SPI can reach a maximum data transfer rate of 10Mbit/s. This is 1.25 MB/s and therefore your expectation is not achievable. In combination with a file system the speed is reduced again. See the following issue #22906. There and in the following issues, there are more measurements.