zephyrproject-rtos/zephyr #29519: kernel: provide alignment variants for k_malloc(), k_mem_pool_malloc(), k_mem_pool_alloc()

cfriedt, 2020-10-24:
**Is your enhancement proposal related to a problem? Please describe.**
While it is not really meaningful to provide allocators for arbitrary alignments (1, 2, 3, .., 42) it should certainly be feasible to adjust the existing allocators so that they are able to select memory blocks to return based on alignment criteria. 

**Describe the solution you'd like**
* modify the allocators in question to accept an alignment parameter and rename
* create wrapper versions of the old allocators (which request alignment of (void *), possibly static inline)
* adjust the primary allocation routine (k_mem_pool_alloc()) to select based on alignment

**Describe alternatives you've considered**

**Additional context**
This will be used for dynamic thread stacks, including #29029

andrewboie, 2020-10-25:
k_mem_pool_* and sys_mem_pool_* are being removed from the kernel.

cfriedt, 2020-10-26:
> k_mem_pool_* and sys_mem_pool_* are being removed from the kernel.

Just saw #28611 - now everything makes sense! Thanks ;-)

PR #30363 (2020-12-27): kernel/k_malloc: provide aligned allocators
kernel/k_malloc: provided aligned allocators

This change adds z_heap_aligned_alloc() and k_aligned_alloc() and changes z_heap_malloc() and k_malloc() to be small wrappers around the aligned variants.

Fixes #29519
commit 05a08a3b (2020-12-27): tests: kernel/k_malloc: tests for k_aligned_alloc

This change adds tests for k_aligned_alloc.

Fixes #29519
commit 135ffaff (2020-12-27): kernel/k_malloc: add k_aligned_alloc

This change adds z_heap_aligned_alloc() and k_aligned_alloc()
and changes z_heap_malloc() and k_malloc() to be small wrappers around
the aligned variants.

Fixes #29519
closed 2020-12-27 by carlescufi