zephyrproject-rtos/zephyr #29530: display: nrf52840: adafruit_2_8_tft_touch_v2 shield not working with nrf-spim driver

nick-kraus, 2020-10-25:
**Describe the bug**
Neither the drivers/display nor the display/lvgl samples work with the default nrf52840dk_nrf52840 devicetree and adafruit_2_8_tft_touch_v2 shield.

With no changes to the dts only a white screen is displayed for either demo. Changing the 'arduino_spi' node to be 'spi1' instead of 'spi3' (but still using the nrf-spim driver, like the 'spi3' node must use) causes the display to show alternating grey and white bars which go up about three quarters of the display.

Changing the 'arduino_spi: &spi1' node to then use the 'nrf-spi' driver instead of the 'nrf-spim' driver resolves the issue, and both samples begin to work.

**To Reproduce**
Steps to reproduce the behavior:
1. west init && west update
2. west build -p -b nrf52840dk_nrf52840 samples/drivers/display/ -- -DSHIELD=adafruit_2_8_tft_touch_v2
3. west flash
4. observe issue
5. make suggested changes above
6. observe working display

**Expected behavior**
The display samples should work without modification on the board combinations suggested by documentation.

**Impact**
Default sample does not work, causing a poor experience when trying to start using a display with the board combinations suggested in the zephyr documentation.

The 'SPI' peripheral of the nrf52840 also doesn't support easyDMA, whereas the 'SPIM' peripheral does, so I would also expect to see diminished performance when having to use the spi driver to drive the display instead of the spim driver.

**Environment (please complete the following information):**
 - OS: Windows 10 19041
 - Toolchain: gcc-arm-embedded 9.3.1
 - Commit SHA: 3c79b565fe

**Additional context**

SPI3 w/ nrf-spim driver:
[image]

SPI1 w/ nrf-spim driver:
[image]

SPI1 w/ nrf-spi driver:
[image]

jfischer-phytec-iot, 2020-11-02:
I can not reproduce it with latest master (b3ff89bd511afef484c193a428633f6a1c866944), please recheck.

nick-kraus, 2020-11-07:
Sorry for the late reply. Got around to checking this again, still does not work on latest master for me (577d588), and also does not work on the commit that you mentioned (b3ff89b), just getting the fully white screen. Just tried it out on both of those commits on a nucleo_f767zi board, which worked no issue.

Looking over my nrf52840 DK, and it is actually a _nrf52840 preview DK_, and it looks like I have a super early version with an engineering sample A nrf52840 chip. Looking through [the errata for this chip](https://infocenter.nordicsemi.com/pdf/nRF52840_Engineering_A_Errata_v1.6.pdf) (PDF warning), and it actually explains my issue. On this silicon rev both the SPIM3 peripheral doesn't work at all, and the SPIM peripheral in general has an issue where it clocks out an extra byte, which is probably what is happening.

So, seems like this is due to me having some extremely early hardware, and probably isn't something that could be recreated on a recent setup. Should be all good to close this issue IMO.

jfischer-no, 2020-11-17:
okay, thanks for response

closed 2020-11-17 by jfischer-no