zephyrproject-rtos/zephyr #29535: kernel:Thread stack frame is not rounded up to arch stack alignment

Teng-Shih-Wei, 2020-10-26:
**Describe the bug**
In macro "Z_STACK_PTR_TO_FRAME", it does not round up the stack frame size to arch 
stack pointer alignment. Therefore, the thread stack size is not rounded up when initializing, 
but it is when restoring the context to run.  The bottom of thread stack is actually over it 
was allocated at first.  

**To Reproduce**
Enable CONFIG_STACK_SENTINEL and error will be triggered

andrewboie, 2020-10-27:
Thanks for the fix....but that's very weird that the size of the frame itself isn't aligned to the stack pointer alignment, since it represents data pre-populated onto the stack. What arch did you see this on?

Teng-Shih-Wei, 2020-10-28:
I found this issue on RISC-V at first. If we enable CONFIG_FPU and CONFIG_FPU_SHARING, the stack frame size is 41*4/41*8 for RV32/RV64. They are not 16-bytes alignment. 

On the other hand, there is an member "soc_context" in stack frame for extension. It would be a potential risk for user if he does not notice it.

Some archs only need 4-byte alignment, they may not have this issue. If any archs need 8/16-bytes alignment, it would be better to add this protection. Therefore, I would like to add it for all archs instead of only RISC-V.

andrewboie, 2020-10-28:
> I found this issue on RISC-V at first. If we enable CONFIG_FPU and CONFIG_FPU_SHARING, the stack frame size is 41x4/41x8 for RV32/RV64. They are not 16-bytes alignment.

So here's the question: *why* is an irregular amount of data being put on the stack? If we are putting data on the stack to instantiate a new thread, how are we getting into a situation where the amount of data pushed onto the stack results in a C ABI violation?

katsuster, 2020-12-11:
@andrewboie
RISC-V standard calling convention requires stack pointer must be 16bytes aligned.
See: https://riscv.org/wp-content/uploads/2015/01/riscv-calling.pdf
18.2  RVG Calling Convention

If users use only C functions compilers keep SP 16bytes aligned, but in z_setup_new_thread(), it sets SP manually.
So Zephyr violates SP alignment rules of RISC-V.

andrewboie, 2020-12-17:
I posted analysis and an alternate approach in the linked PR. Long story short, your stack object is itself mis-aligned as its base address and size are only aligned to Z_RISCV_PMP_ALIGN.

andrewboie, 2020-12-17:
@katsuster @Teng-Shih-Wei ^^

katsuster, 2021-04-09:
Reopened...
We need to solve this issue. Current RV32 Zephyr printf() outputs very strange values by this problem.

```
===================================================================
START - test_pi
Pi calculation OK after 50 (high) + 118 (low) tests (computed -26815615863448970000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.000000)
Pi calculation OK after 150 (high) + 360 (low) tests (computed -26815615863448970000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.000000)
Pi calculation OK after 250 (high) + 595 (low) tests (computed -26815615863448970000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.000000)
Pi calculation OK after 350 (high) + 828 (low) tests (computed -26815615863448970000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.000000)
Pi calculation OK after 450 (high) + 1054 (low) tests (computed -26815615863448970000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.000000)
 PASS - test_pi in 12.306 seconds
```

commit d109805c (2021-06-11): RISC-V: Round up pre-populated stack frame to arch stack alignment

The stack frame size, used for context switch, is rounded up to 16-bytes
alignment. Therefore, we need round down the pointer of top of the
pre-populated stack frame so that the preserved stack frame size is also
rounded up to 16-bytes alignment.

Fixes #29535
closed 2021-06-11 by carlescufi