zephyrproject-rtos/zephyr #29538: LwM2M client doesn't work on disco_l475_iot1

panekk, 2020-10-26:
**Describe the bug**
Zephyr's LwM2M client uses `recvfrom` function with non-NULL `from` and `fromlen` arguments, but `eswifi_socket_recvfrom()` doesn't support them. Is there any chance to run the LwM2M client on B-L475E-IOT01A board?

**To Reproduce**
Run LwM2M client on B-L475E-IOT01A and try to register to the LwM2M Server.

**Expected behavior**
Successful registration to the server. 

**Impact**
LwM2M client is unusable when eswifi modem is used.

**Logs and console output**
```
[00:00:04.848,000] <inf> net_lwm2m_rd_client: RD Client started with endpoint 'zephyr-test' with client lifetime 30
[00:00:05.412,000] <err> net_lwm2m_engine: Error reading response: 134
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK
 - Version used: Zephyr OS build zephyr-v2.3.0-599-g1105aa521c29

jukkar, 2020-10-26:
Hmm, I wonder why eswifi does not support this.

cc: @rlubos & @loicpoulain

rlubos, 2020-10-26:
Well I'm a bit confused with this driver in general, since it seems to implement both, net offloading and socket offloading. Which one is used in this case?

As for socket offloading approach, it's clear that the `recvfrom()` implementation is incomplete:
https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/wifi/eswifi/eswifi_socket_offload.c#L349

panekk, 2020-10-26:
`CONFIG_WIFI_ESWIFI` selects `NET_SOCKETS_OFFLOAD` by default, so I wonder if it's even possible to disable the offloading. Setting `NET_SOCKETS_OFFLOAD`, `NET_OFFLOAD` and `WIFI_OFFLOAD` doesn't help anyway, with warnings printed for the second two (`(...) was assigned the value 'n' but got the value 'y'`). It isn't possible to disable the offloading using menuconfig too.

rlubos, 2020-10-28:
@nashif I'm not the one who should work on this one, It's not an LwM2M issue but a faulty/incomplete wifi driver implementation.

@panekk If eswifi does `select NET_SOCKETS_OFFLOAD` it means it uses socket offloading, hence the above answer explains the behaviour (the `recvfrom` implementation is incomplete). Unfortunately it seems the original author is not reachable through github anymore, I'm not sure who should be notified about this.
BTW, if some module uses `select` on certain symbol in Kconfig, it's not possible to "deselect" it.

loicpoulain, 2020-11-02:
Going to have a look at this by the end of the week

loicpoulain, 2020-11-09:
So, looking at eswifi behavior, to support that we need to handle 'UDP SVR' events which are on the below form:
[UDP SVR] Accepted 192.168.1.23:4242[EOMA]

We need to save ip/port so that it can be passed back in recvfrom. Seems only one UDP client can be handled at a time, we need to stop/restart UDP 'accept' after each client... will try to implement something.

closed 2021-01-23 by github-actions[bot]