zephyrproject-rtos/zephyr #29541: CONFIG_THREAD_LOCAL_STORAGE=y build fails with ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb

pfalcon, 2020-10-26:
ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb is one of the supported toolchains to build Zephyr (for ARM architecture). After https://github.com/zephyrproject-rtos/zephyr/pull/28982 was merged, and with CONFIG_THREAD_LOCAL_STORAGE enabled (which it is, for one sanitycheck configuration of tests/kernel/common), the build fails with:

```
[ 87%] Linking C executable app_smem_unaligned_prebuilt.elf
/home/buildslave/srv/toolchain/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /home/buildslave/srv/toolchain/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/thumb/v7e-m/nofp/libgcc.a(emutls.o): in function `emutls_alloc':
emutls.c:(.text+0x5a): undefined reference to `abort'
/home/buildslave/srv/toolchain/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /home/buildslave/srv/toolchain/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/thumb/v7e-m/nofp/libgcc.a(emutls.o): in function `__emutls_get_address':
emutls.c:(.text+0x7a): undefined reference to `abort'
collect2: error: ld returned 1 exit status
make[2]: *** [zephyr/app_smem_unaligned_prebuilt.elf] Error 1
make[1]: *** [zephyr/CMakeFiles/app_smem_unaligned_prebuilt.dir/all] Error 2
make: *** [all] Error 2

INFO    - /home/buildslave/srv/zephyr/master/gnuarmemb/frdm_k64f/frdm_k64f/tests/kernel/common/kernel.common.tls/build.log
```

Full build log: https://ci.linaro.org/job/zephyr-upstream/PLATFORM=frdm_k64f,ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb,label=docker-xenial-amd64-13/5680/console

pfalcon, 2020-10-26:
@galak: FYI

pfalcon, 2020-10-26:
This error doesn't happen if built with CONFIG_NEWLIB_LIBC=y, which leads us to the patch https://github.com/zephyrproject-rtos/zephyr/pull/29542.

Note that I didn't try to investigate why emutls.c "the emulation of thread local storage using malloc()" https://github.com/gcc-mirror/gcc/blob/master/libgcc/emutls.c is being used. @galak, let me know if you think that should be investigated.

galak, 2020-10-27:
@dcpleung fyi

galak, 2020-10-27:
since this is the case w/minlibc that fails, its odd to me that this works ok w/Zephyr SDK & fails w/gnuarmemb.

pfalcon, 2020-10-27:
Apparently, some implicitly linked libs included with Zephyr SDK provide abort().

galak, 2020-10-27:
> Apparently, some implicitly linked libs included with Zephyr SDK provide abort().

looking at the error message in more detail, I see this is in libgcc, so that partially explains it.

galak, 2020-10-27:
Based on https://github.com/ARM-software/toolchain-gnu-bare-metal/blob/365499ead3c18e9c1a70880c34d8efe08549df3c/build-toolchain.sh#L423 we should probably change things so TLS is only supported w/Zephyr toolchain.

dcpleung, 2020-10-27:
Submitted #29591 to limit the kconfig option to Zephyr SDK only. Hopefully that fixes the issue.

pfalcon, 2020-10-27:
> we should probably change things so TLS is only supported w/Zephyr toolchain.

Right. But that's orthogonal to the missing abort() in minlibc, PR #29542.

andrewboie, 2020-10-27:
It's not clear to me the root cause here. Does this toolchain really not support TLS at all? Or is this just some reference to abort() from libgcc that is simply fixed by https://github.com/zephyrproject-rtos/zephyr/pull/29542

dcpleung, 2020-10-27:
According to the link from @galak, it looks like the toolchain is built without TLS support (as configured with `--disable-tls`).

PR #29542 (2020-10-28): libc: minimal: Implement abort().
abort() is an important runtime function, oftentimes used to signal
abnormal execution conditions in generic applications. Worse, they
may be used under such circumstances in e.g. compiler support
libraries, in which case lack of implementation of this function
will lead to link error.

Fixes: #29541
commit 56522851 (2020-10-28): libc: minimal: Implement abort().

abort() is an important runtime function, oftentimes used to signal
abnormal execution conditions in generic applications. Worse, they
may be used under such circumstances in e.g. compiler support
libraries, in which case lack of implementation of this function
will lead to link error.

Fixes: #29541
PR #29591 (2020-10-30): kernel: limit thread local storage to Zephyr SDK
Toolchains other than Zephyr SDK may not support generating
code with thread local storage. So limit TLS to Zephyr SDK
for now, and only enable TLS on other toolchains as needed.

Fixes #29541
commit 2d152ab0 (2020-10-30): kernel: limit thread local storage to Zephyr SDK

Toolchains other than Zephyr SDK may not support generating
code with thread local storage. So limit TLS to Zephyr SDK
for now, and only enable TLS on other toolchains as needed.

Fixes #29541
closed 2020-10-28 by carlescufi