zephyrproject-rtos/zephyr #29544: Bluetooth: Mesh: Friend node unable relay message for lpn

LingaoM, 2020-10-26:
As defined by Mesh Specification 1.0.1 Section  3.6.6.2 Friendship security.

`The Low Power node then decides to send two messages: OutMsg1 and OutMsg2. OutMsg1 is sentsecured using the friend security credentials and therefore only the Friend node will receive and relay thismessage. When the Friend node relays OutMsg1, the message will be retransmitted using the mastersecurity credentials. OutMsg2 is sent secured using the master security credentials and therefore theFriend node and any other Relay node within range of the Low Power node can relay the message.OutMsg2, when it is relayed, will be retransmitted using the master security credentials. `

If low power node publish with unicast address other than friend address, the friend node not relayed this message, no matter friend cred used.

https://github.com/zephyrproject-rtos/zephyr/blob/ba9fd65a4bfaeb193f84b52f3e0bf09e1ac2adfc/subsys/bluetooth/mesh/net.c#L797-L805

Another problem in `bt_mesh_net_relay`
https://github.com/zephyrproject-rtos/zephyr/blob/ba9fd65a4bfaeb193f84b52f3e0bf09e1ac2adfc/subsys/bluetooth/mesh/net.c#L614-L618

This implementation obviously prohibit relay message for low power node, even local is friend node.

trond-snekvik, 2020-10-26:
I think I need a bit more details here. The check in the first code snippet will relay anything that has a group destination address or a unicast destination other than itself or the LPN. If LPN sends a message to a random unicast address, the friend node will check this, and find that `!BT_MESH_ADDR_IS_UNICAST(rx.ctx.recv_dst)` is `!true` and `(!rx.local_match && !rx.friend_match)` is `(!false && !false)` => `true`, then relay the message, using `&rx->sub->keys[SUBNET_KEY_TX_IDX(rx->sub)].msg` for encryption. The source address of the message never comes into play. What am I missing?

The friend's relay to the LPN is done in transport, and happens before the second snippet:

https://github.com/zephyrproject-rtos/zephyr/blob/ba9fd65a4bfaeb193f84b52f3e0bf09e1ac2adfc/subsys/bluetooth/mesh/transport.c#L1558-L1565

LingaoM, 2020-10-26:
You may have misunderstood what I mean, I need to emphasize that lpn can publish messages, for example with unicast address other than friend node and use friend cred, and this message can be relayed by friends, as defined by Mesh Specification 1.0.1 Section 3.6.6.2 Friendship security.

`is (!false && !true)`

 Even if the bt_mesh_net_relay function is called, the message may not be relayed by friend, because the local node may not support relay or proxy or disabled.

LingaoM, 2020-10-26:
> The friend's relay to the LPN is done in transport, and happens before the second snippet:
> 
> https://github.com/zephyrproject-rtos/zephyr/blob/ba9fd65a4bfaeb193f84b52f3e0bf09e1ac2adfc/subsys/bluetooth/mesh/transport.c#L1558-L1565

What I want to saying is the problem of the message being relayed by the friend node, not the problem of the friend node being relayed to lpn.

LingaoM, 2020-10-27:
> I think I need a bit more details here. The check in the first code snippet will relay anything that has a group destination address or a unicast destination other than itself or the LPN. If LPN [...]

@trond-snekvik  Sorry to cause you trouble, I confuse friend_match and friend_cred. But I still think there are some problems in handling relay. I tried to solve this problem through PR. Please review.

PR #29580 (2020-10-29): Bluetooth: Mesh: Fixes Relay to frnd_cred message
If low power node publish with unicast address other
than friend address, the friend node should relayed
this message to all network interface.

Fixes: #29544
commit 1e9bd13c (2020-10-29): Bluetooth: Mesh: Fixes Relay to frnd_cred message

If low power node publish with unicast address other
than friend address, the friend node should relayed
this message to all network interface.

When Proxy feature enabled, message from gatt bearer should
relay to adv bearer even relay feature disabled.

Fixes: #29544
closed 2020-10-29 by jhedberg