zephyrproject-rtos/zephyr #29545: samples: tfm_integration: tfm_ipc: No module named 'cryptography.hazmat.primitives.asymmetric.ed25519'

ABOSTM, 2020-10-26:
**Describe the bug**
While compiling "samples/tfm_integration/tfm_ipc" on test bench,
there is a compilation error linked to pyhon import error:
ModuleNotFoundError: No module named 'cryptography.hazmat.primitives.asymmetric.ed25519'

**To Reproduce**
Steps to reproduce the behavior:
1. sanitycheck -N --device-testing --hardware-map ../map.yaml -c -p nucleo_l552ze_q_ns - T samples/tfm_integration/tfm_ipc
2. See error

**Expected behavior**
Compilation successful 

**Logs and console output**
```
[560/582] Generating tfm_s_signed.bin
FAILED: bl2/ext/mcuboot/tfm_s_signed.bin
cd /local/ztb/logs/201023/log_job_STM32_CUBE_UPDATES_PR75_d3c31976b0_1750/nucleo_l552ze_q_ns/samples/tfm_integration/tfm_ipc/sample.tfm_ipc/tfm/lib/ext/mcuboot-src/scripts && /usr/bin/python3 /local/ztb/zephyrproject/modules/tee/tfm/trusted-firmware-m/bl2/ext/mcuboot/scripts/wrapper/wrapper.py -v 1.1.0 --layout /local/ztb/logs/201023/log_job_STM32_CUBE_UPDATES_PR75_d3c31976b0_1750/nucleo_l552ze_q_ns/samples/tfm_integration/tfm_ipc/sample.tfm_ipc/tfm/bl2/ext/mcuboot/CMakeFiles/signing_layout_s.dir/./signing_layout_s.o -k /local/ztb/zephyrproject/modules/tee/tfm/trusted-firmware-m/bl2/ext/mcuboot/root-RSA-3072.pem --public-key-format full --align 1 --pad --pad-header -H 0x400 -s auto -d "(0,0.0.0+0)" --overwrite-only  /local/ztb/logs/201023/log_job_STM32_CUBE_UPDATES_PR75_d3c31976b0_1750/nucleo_l552ze_q_ns/samples/tfm_integration/tfm_ipc/sample.tfm_ipc/tfm/bin/tfm_s.bin /local/ztb/logs/201023/log_job_STM32_CUBE_UPDATES_PR75_d3c31976b0_1750/nucleo_l552ze_q_ns/samples/tfm_integration/tfm_ipc/sample.tfm_ipc/tfm/bl2/ext/mcuboot/tfm_s_signed.bin && /local/home/zephyrtb/.local/lib/python3.6/site-packages/cmake/data/bin/cmake -E copy /local/ztb/logs/201023/log_job_STM32_CUBE_UPDATES_PR75_d3c31976b0_1750/nucleo_l552ze_q_ns/samples/tfm_integration/tfm_ipc/sample.tfm_ipc/tfm/bl2/ext/mcuboot/tfm_s_signed.bin /local/ztb/logs/201023/log_job_STM32_CUBE_UPDATES_PR75_d3c31976b0_1750/nucleo_l552ze_q_ns/samples/tfm_integration/tfm_ipc/sample.tfm_ipc/tfm/bin
Traceback (most recent call last):
  File "/local/ztb/zephyrproject/modules/tee/tfm/trusted-firmware-m/bl2/ext/mcuboot/scripts/wrapper/wrapper.py", line 21, in <module>
    import imgtool.main
  File "/local/ztb/logs/201023/log_job_STM32_CUBE_UPDATES_PR75_d3c31976b0_1750/nucleo_l552ze_q_ns/samples/tfm_integration/tfm_ipc/sample.tfm_ipc/tfm/lib/ext/mcuboot-src/scripts/imgtool/main.py", line 21, in <module>
    import imgtool.keys as keys
  File "/local/ztb/logs/201023/log_job_STM32_CUBE_UPDATES_PR75_d3c31976b0_1750/nucleo_l552ze_q_ns/samples/tfm_integration/tfm_ipc/sample.tfm_ipc/tfm/lib/ext/mcuboot-src/scripts/imgtool/keys/__init__.py", line 25, in <module>
    from cryptography.hazmat.primitives.asymmetric.ed25519 import (
ModuleNotFoundError: No module named 'cryptography.hazmat.primitives.asymmetric.ed25519'

```

**Environment (please complete the following information):**
 - OS:  Linux
 - Toolchain Zephyr SDK
 - Commit SHA :Zephyr: ba9fd65a4bfaeb193f84b52f3e0bf09e1ac2adfc
tfm: bb15a511a509c81135af61e02f9f4bfaf86d7fde

**Additional context**
Add any other context about the problem here.

omitride, 2021-01-08:
I have the exact same error message.
I suspect that the build scripts of the tfm module are not using the correct environment for Python if the user has a [virtual environment](https://docs.python.org/3/tutorial/venv.html) for Python.

In my case, the `cryptography` Python module is installed in the virtual environment I am using for Zephyr, and not in the "global" one.

I can see from the west build output that the generation of the signed.bin files are using the full path of the "global" Python interpreter instead of the one of my virtual environment (or symply "python" instead of a full path).

For debugging purpose, I've forced the value of `PYTHON_EXECUTABLE` to just "python" in "trusted-firmware-m/toolchain_GNUARM.cmake" so that it points to the interpreter of my virtual environment.
In that case the sample builds correctly (`west build samples/tfm_integration/tfm_ipc -b nucleo_l552ze_q_ns -p`).

carlescufi, 2021-02-09:
@ioannisg or @oyvindronningstad any thoughts on this one?

ioannisg, 2021-02-10:
Can't reproduce this , maybe @oyvindronningstad or @microbuilder  have encountered this one.

oyvindronningstad, 2021-02-10:
Looks like Ed25519 was added in 2.6 - in February 2019 (https://cryptography.io/en/latest/changelog.html#v2-6). Is it possible that you just need to update your `cryptography` pip module, @ABOSTM?

ioannisg, 2021-02-10:
That's what I also think, but shouldn't this be an added "requirement" for TF-M @oyvindronningstad ?

oyvindronningstad, 2021-02-10:
The failure above happens in mcuboot (imgtool), which does specify >=2.6 in its requirements.txt. tfm does not, but I don't know what they use it for.

ABOSTM, 2021-03-04:
Hi @oyvindronningstad @ioannisg,
I am sorry, I totally missed that point.
I just retested with master branch and west update,
and I could not reproduce the issue.
Note: my current cryptography version: 3.3.1

closed 2021-03-04 by ABOSTM

nxf58843, 2021-07-09:
For anyone else who gets to this issue because python cannot find 'cryptography.hazmat.primitives.asymmetric.ed25519'
I ran into this and finally realized that pip was only installing for python 2.7.  I had to install cryptography with pip3 to fix it.
`pip3 install cryptography --force-reinstall`

erwango, 2022-01-03:
> For anyone else who gets to this issue because python cannot find 'cryptography.hazmat.primitives.asymmetric.ed25519' I ran into this and finally realized that pip was only installing for python 2.7. I had to install cryptography with pip3 to fix it. `pip3 install cryptography --force-reinstall`

Didn't work for me. Still investigating.

erwango, 2022-01-03:
I suspect there is some dependency/versioning wrongly handled in cryptography module that might be due to the versioning that went from 3.4.0 to 35.0.0 recently (sept' 2021), cf https://pypi.org/project/cryptography/#history.