zephyrproject-rtos/zephyr #29566: Improve West support for MCUboot image encryption

hjuul, 2020-10-27:
After #27688 and #27818 it has become very convenient to sign MCUboot images: The key file is configurable through Kconfig and the build system automatically provides signed and unsigned binaries.

Image _encryption_ on the other hand is not so convenient. On the application side it can be achieved by providing the `-E` flag to `CONFIG_MCUBOOT_EXTRA_IMGTOOL_ARGS`, but then you only get the encrypted image, not a signed unencrypted image suitable for direct flashing.
Also, on the MCUboot side the private key for encryption is hard coded in `mcuboot/boot/zephyr/keys.c`. I'm striving to keep my own code outside the Zephyr tree, so that means I need a fork of MCUboot repo to replace the key.

**Describe the solution you'd like**
There are probably multiple ways this can be done, this is just a suggestion.

For the application _and_ MCUboot:
- Kconfig options to enable encryption and select encryption type
- A Kconfig option to specify the path to the encryption key file (similar to `CONFIG_MCUBOOT_SIGNATURE_KEY_FILE`)

When the application is built, West generates `zephyr.signed.encrypted.[bin|hex]` (in addition to the already existing `zephyr.*` and `zephyr.signed.*`).

When MCUboot is built, West uses `imgtool` to extract the required keys and generate a `keys.c` in the build directory.

henrikbrixandersen, 2022-11-02:
@hjuul Seeing that https://github.com/zephyrproject-rtos/zephyr/pull/37568 was merged, can this be closed?

hjuul, 2022-11-07:
@henrikbrixandersen , #37568 only adds encrypted binaries for the application. It does not handle encryption keys for MCUboot. So I suppose I still need to fork MCUboot repo to add my own keys, which is inconvenient. I prefer keeping this issue open.

hjuul, 2024-04-05:
I see that MCUboot has added a Kconfig option `BOOT_ENCRYPTION_KEY_FILE` now. That solves the last part of this issue, so I'm closing it.

closed 2024-04-05 by hjuul