zephyrproject-rtos/zephyr #29567: Using openthread based echo_lient and lwm2m_client cannot ping the external network address

IHIDCHAOS, 2020-10-27:
**Describe the bug**
I have an openwrt-based border router with Internet access.

There is also nrf52840 dk with panid and channel set up and added to the border router network, thread role is router.

When I use echo_lient or lwm2m_client, I can only ping the ip address of the wpan0 interface of the border router, but cannot ping the external public network address, such as google dns.

But when I use the thread client officially provided by openthread or nordic, I can successfully ping Google dns.

Do I need other additional configuration?

My environment:

## **openwrt:**

**border router:**

```
panid 0xabcd

channel 11
```

I don't have a native ipv6 environment, so I configured nat64 and dns64.

**nat64**: tayga

```
option prefix'fd00:470:1f09:97c5::/96'
```

**dns64**: totd

```
prefix fd00:470:1f09:97c5::
```

**The internet**:

```
root@Hoo:~# ifconfig wpan0
wpan0 Link encap:UNSPEC HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet6 addr: fd11:1111:2222:0:c15e:e7aa:f661:82b6/64 Scope:Global
          inet6 addr: fd11:1111:2222::ff:fe00:fc00/64 Scope:Global
          inet6 addr: fd11:1111:2222::ff:fe00:e400/64 Scope:Global
          inet6 addr: fd11:22::1436:b48f:fea3:5b61/64 Scope:Global
          inet6 addr: fe80::145a:e58b:d00b:bf72/64 Scope:Link
          UP POINTOPOINT RUNNING NOARP MULTICAST MTU:1500 Metric:1
          RX packets:146 errors:0 dropped:2 overruns:0 frame:0
          TX packets:267 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:500
          RX bytes: 13653 (13.3 KiB) TX bytes: 25040 (24.4 KiB)

```

```
root@Hoo:~# ping fd00:470:1f09:97c5::0808:0808
PING fd00:470:1f09:97c5::0808:0808 (fd00:470:1f09:97c5::808:808): 56 data bytes
64 bytes from fd00:470:1f09:97c5::808:808: seq=0 ttl=113 time=60.831 ms
64 bytes from fd00:470:1f09:97c5::808:808: seq=1 ttl=113 time=73.499 ms
^C
--- fd00:470:1f09:97c5::0808:0808 ping statistics ---
2 packets transmitted, 2 packets received, 0% packet loss
round-trip min/avg/max = 60.831/67.165/73.499 ms
root@Hoo:~#
```

**To Reproduce**
Steps to reproduce the behavior:
1. cd sample/net/lwm2m
2. mkdir build && cd build
3. cmake -GNinja -DBOARD=nrf52840dk_nrf52840 -DOVERLAY_CONFIG=overlay-ot.conf ..
4. ninja
5. west flash 

**Expected behavior**
i wang to ping from nrf52840 board to pulbic web such as google dns

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**

when using zephyr and lwm2m_client
```
▒*** Booting Zephyr OS build zephyr-v2.4.0-1017-g060e00d9c514  ***

[00:00:00.283,355] <dbg> net_lwm2m_obj_device.device_create: Create LWM2M device instance: 0
[00:00:00.283,386] <dbg> net_lwm2m_obj_conn_mon.connmon_create: Create LWM2M connectivity monitoring instance: 0
[00:00:00.283,416] <dbg> net_lwm2m_obj_firmware.firmware_create: Create LWM2M firmware instance: 0
[00:00:00.283,416] <dbg> net_lwm2m_obj_location.location_create: Create Location instance: 0
[00:00:00.283,447] <inf> net_config: Initializing network
[00:00:00.283,599] <inf> net_l2_openthread: State changed! Flags: 0x00000001 Current role: 1
[00:00:00.283,660] <inf> net_lwm2m_client_app: Run LWM2M client
[00:00:00.283,721] <dbg> net_lwm2m_engine.lwm2m_engine_set: path:0/0/2, value:0x200186f7, len:1
[00:00:00.283,752] <dbg> net_lwm2m_engine.lwm2m_engine_set: path:0/0/1, value:0x200186f7, len:1
[00:00:00.283,782] <dbg> net_lwm2m_engine.lwm2m_engine_create_obj_inst: path:0/1
[00:00:00.283,843] <dbg> net_lwm2m_obj_security.security_create: Create LWM2M security instance: 1
[00:00:00.284,210] <dbg> net_lwm2m_engine.lwm2m_engine_create_obj_inst: path:3303/0
[00:00:00.284,271] <dbg> net_ipso_temp_sensor.temp_sensor_create: Create IPSO Temperature Sensor instance: 0
[00:00:00.284,301] <dbg> net_lwm2m_engine.lwm2m_engine_create_obj_inst: path:3311/0
[00:00:00.284,332] <dbg> net_ipso_light_control.light_control_create: Create IPSO Light Control instance: 0
[00:00:00.284,393] <dbg> net_lwm2m_engine.lwm2m_engine_create_obj_inst: path:3340/0
[00:00:00.284,454] <dbg> net_ipso_timer.timer_create: Create IPSO Timer instance: 0
[00:00:00.284,576] <inf> net_lwm2m_rd_client: Start LWM2M Client: 000003740521568;13849909253
[00:00:00.414,855] <inf> net_l2_openthread: State changed! Flags: 0x000010e4 Current role: 3
[00:00:00.432,556] <inf> net_l2_openthread: State changed! Flags: 0x00000200 Current role: 3
[00:00:01.283,294] <dbg> net_lwm2m_engine.lwm2m_engine_get: path:0/0/1, buf:0x20017c07, buflen:1
[00:00:01.283,325] <inf> net_lwm2m_rd_client: Bootstrap started with endpoint '000003740521568;13849909253' with client lifetime 0
[00:00:01.283,355] <dbg> net_lwm2m_engine.lwm2m_parse_peerinfo: Parse url: coap://[fd00:470:1f09:97c5::b7e6:2827]
[00:00:01.285,339] <dbg> net_lwm2m_rd_client.sm_do_bootstrap_reg: Register ID with bootstrap server as 'ep=000003740521568;13849909253'
[00:00:01.285,675] <err> net_otPlat_radio: Error while calling otIp6Send
[00:00:03.630,584] <inf> net_lwm2m_engine: Resending message: 0x200095ac
[00:00:03.630,920] <err> net_otPlat_radio: Error while calling otIp6Send
[00:00:08.318,878] <inf> net_lwm2m_engine: Resending message: 0x200095ac
[00:00:08.319,213] <err> net_otPlat_radio: Error while calling otIp6Send
```

```
uart:~$ ot state
router
Done
uart:~$ ot ipaddr
fd11:1111:2222:0:0:ff:fe00:9c00
fdde:ad00:beef:0:0:0:0:1
fd11:1111:2222:0:62bc:3e14:49bf:1abe
fe80:0:0:0:e43f:9eec:68b1:4547
Done
uart:~$ kernel version
Zephyr version 2.4.99
uart:~$ net ping fd00:470:1f09:97c5::0808:0808
PING fd00:470:1f09:97c5::0808:0808
Ping timeout
```

when using nordic offical client:
```
> version
OPENTHREAD/20191113-00534-gc6a258e3; NRF52840; Apr  5 2020 21:51:18
Done
> state
disabled
Done
> panid 0xabcd
Done
> channel 11
Done
> ifconfig up
Done
> thread start
Done
> state
child
Done
> ping fd00:470:1f09:97c5::0808:0808
Done
> 16 bytes from fd00:470:1f09:97c5:0:0:808:808: icmp_seq=1 hlim=112 time=47ms
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: commit 060e00d9c514a8df0dc3e8e7c271631a561a7726 (HEAD -> master, origin/master)

**Additional context**
Add any other context about the problem here.

rlubos, 2020-10-27:
It seems there's no global IPv6 address configured on your Thread device. Try enabling the following:
* `CONFIG_OPENTHREAD_SLAAC=y` for IPv6 autoconfiguration based on the global prefix distributed by the BR,
* `CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=6` to increase maximum IPv6 address per Zephyr network interface. The default value is so finetuned that it was needed in my case.

This configuration works for me, I was able to ping Google's DNS server with the echo sample (I've used echo_server to avoid additional noise from the app):
 ```
uart:~$ net ping 64:ff9b::0808:0808
PING 64:ff9b::0808:0808
8 bytes from 64:ff9b::808:808 to fd11:22::e5c4:9c5b:8473:bb32: icmp_seq=0 ttl=124 rssi=0 time=40 ms
8 bytes from 64:ff9b::808:808 to fd11:22::e5c4:9c5b:8473:bb32: icmp_seq=1 ttl=124 rssi=0 time=41 ms
8 bytes from 64:ff9b::808:808 to fd11:22::e5c4:9c5b:8473:bb32: icmp_seq=2 ttl=124 rssi=0 time=39 ms
```

IHIDCHAOS, 2020-10-28:
According to the configuration you mentioned, I used echo_server and added the two configuration into overlay-ot.conf, and tried again. But it still doesn't work.

It seems that Zephyr still cannot get On-Mesh-Prefix Addres from OTBR.

**OTBR:**
[image]

**Nordic Client: success**
[image]

**Zephyr echo_server: failed**
[image]

rlubos, 2020-10-28:
Did you run a clean build? I don't know why, but OpenThread does not seem to catch up with changes in config files if there's already an old build present.

It might also be the case, that the SLAAC option is disabled for the prefix you distribute (I don't know your BR configuration, in OTBR it's enabled), so you might need to use DHCPv6 instead (enabled with `CONFIG_OPENTHREAD_DHCP6_SERVER=y`). But I've never used this variant to be honest.

IHIDCHAOS, 2020-10-29:
Thank you! The problem is solved.

When I execute the following command
```
cd build
rm -rf *
cmake -GNinja -DBOARD=nrf52840dk_nrf52840 -DOVERLAY_CONFIG=overlay-ot.conf ..
ninja
west flash
```
it became normal. 

And to me, enabling
```
CONFIG_OPENTHREAD_DHCP6_SERVER=y
```
alone without SLAAC seems to have no effect.

In addition I found that in my case, 
```
CONFIG_OPENTHREAD_SLAAC=y
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=6
```
must be enabled together, otherwise my board is only visible to get the address, but can not access the Internet.

**when all default:**

no ip from otbr

[image]

**when only enable slaac:**

got ip but no internet access

[image]

**when enable both:**

got ip and ping ok

[image]

rlubos, 2020-10-29:
Yes, you need both options. `CONFIG_OPENTHREAD_SLAAC=y` is needed to allow OpenThread itself to configure the global address. But  OpenThread needs also to register the address on a network interface, that's why you need to increase 
 `CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT`, otherwise the address fails to register and Zephyr cannot use it.

It might also be wise to disable statically configured IPv6 address if you don't need it with `CONFIG_NET_CONFIG_MY_IPV6_ADDR=""`. We only set it in the samples to make them work out of the box with each other, but in general this shall not be done (and the truth is it's kind of hacky solution for OpenThread). 

Can we close the issue then?

IHIDCHAOS, 2020-10-29:
Yes, thank you very much for your help!

closed 2020-10-29 by IHIDCHAOS