zephyrproject-rtos/zephyr #29574: question: about CONFIG_NET_BUF_POOL_USAGE

ehudmarvell, 2020-10-27:
Hi,

I am using CONFIG_NET_BUF_POOL_USAGE and saw that increasing and decreasing "avail_count" is done without atomic variable and without any locks.

for example:
```
#if defined(CONFIG_NET_BUF_POOL_USAGE)
	pool->avail_count--;
	__ASSERT_NO_MSG(pool->avail_count >= 0);
#endif
```

Can you explain how can I be sure that avail_count is valid, is there any protections on it that I miss?

Thanks,
Ehud

jukkar, 2020-10-27:
The `avail_count` is currently used as a statistics variable to show some information in net-shell. This could be converted to be atomic_t if needed. Patches are welcome if you want to improve the code.

ehudmarvell, 2020-10-27:
@jukkar 
see my patch at:
https://github.com/zephyrproject-rtos/zephyr/compare/master...ehudmarvell:fix_29574?expand=1

closed 2020-10-28 by ehudmarvell