zephyrproject-rtos/zephyr #29581: LoRaWAN sample for 96b_wistrio - Tx timeout

vtothsvk, 2020-10-27:
Hi!

I ran into the following issue trying to run the lorawan sample on the 96b_wistrio sample: I get a Tx timeout error every time a join request or message is sent. Tried to join using ABP as well, then the join request succeeds, however any messages sent still result in a Tx timeout.

sample output using OTA:

```
*** Booting Zephyr OS build zephyr-v2.4.0-1018-g1b3abc096cc4  ***<\r><\n>
[00:00:00.200,000] <27>[0m<inf> sx1276: SX1276 Version:12 found<27>[0m<\r><\n>
[00:00:00.263,000] <27>[0m<inf> lorawan_class_a: Joining network over OTAA<27>[0m<\r><\n>
[00:00:04.314,000] <27>[1;31m<err> lorawan: MlmeConfirm failed : Tx timeout<27>[0m<\r><\n>
[00:00:04.314,000] <27>[1;31m<err> lorawan_class_a: lorawan_join_network failed: -116<27>[0m<\r><\n>
```

In case of the OTA I only changed the EUIs and the App key, rest is the original sample code. The curious thing is that the join request is actually sent and received by the gateway, even though a Tx timeout.

[image]

In the latter run I used the following code to join using ABP:

```
join_cfg.mode = LORAWAN_ACT_ABP;
join_cfg.dev_eui = dev_eui;
join_cfg.abp.app_eui = join_eui;
join_cfg.abp.app_skey = apps_key;
join_cfg.abp.nwk_skey = nws_key;
join_cfg.abp.dev_addr = dev_addr;

LOG_INF("Joining network over ABP");
ret = lorawan_join(&join_cfg);
if (ret < 0) {
	LOG_ERR("lorawan_join_network failed: %d", ret);
	return;
}
```

output using ABP:

```
*** Booting Zephyr OS build zephyr-v2.4.0-1018-g1b3abc096cc4  ***<\r><\n>
[00:00:00.200,000] <27>[0m<inf> sx1276: SX1276 Version:12 found<27>[0m<\r><\n>
[00:00:00.263,000] <27>[0m<inf> lorawan_class_a: Joining network over ABP<27>[0m<\r><\n>
[00:00:00.263,000] <27>[0m<inf> lorawan_class_a: Sending data...<27>[0m<\r><\n>
[00:00:04.278,000] <27>[1;31m<err> lorawan: McpsRequest failed : Tx timeout<27>[0m<\r><\n>
[00:00:04.278,000] <27>[1;31m<err> lorawan_class_a: lorawan_send failed: -116<27>[0m<\r><\n>
```

In spite of getting the Tx timeout error again, the message is sent and received by the gateway as shown in the app data stream.

[image]

I'm stuck on this one since the weekend and really starting to pull my hair out. Try to search what the error actually means and from what I've read it's supposed to be caused when the radio chip is not responding, what should not be the case here, since the data is actually being sent. Moreover the rak811 has the mcu and the sx1276 integrated so I doubt a corrupted connection is the fault. 

Any help would be much appreciated.

Mani-Sadhasivam, 2020-11-01:
@vtothsvk Did you set the region correctly in `prj.conf` file? By default, the region used is `CONFIG_LORAMAC_REGION_IN865`

vtothsvk, 2020-11-01:
> @vtothsvk Did you set the region correctly in `prj.conf` file? By default, the region used is `CONFIG_LORAMAC_REGION_IN865`

Yes, using EU433

vtothsvk, 2020-11-02:
Moreover, I intentionally tried other regions as well, and they had no effect on the Tx timeout whatsoever.

Mani-Sadhasivam, 2020-11-02:
@vtothsvk I just tried the zephyr master on my Wistrio board and didn't get `Tx Timeout` (even without a gateway as expected):

```
*** Booting Zephyr OS build zephyr-v2.4.0-1160-g196ae3506fed  ***
[00:00:00.200,000] <inf> sx1276: SX1276 Version:12 found
[00:00:00.283,000] <inf> lorawan_class_a: Joining network over OTAA
[00:00:07.882,000] <err> lorawan: MlmeConfirm failed : Rx 2 timeout
[00:00:07.882,000] <err> lorawan_class_a: lorawan_join_network failed: -116
```

Can you check the power supply of your board? I'm powering it via Micro USB socket.

adityaxavier, 2020-11-03:
Hi @Mani-Sadhasivam I believe the issue of Join Failure comes in due to DutyCycle Issue with Region IN865 as per LoRaMac-Node change log.. I just did a PR with the updated 4.4.5 which has fixed it for me.
Please Review

KwonTae-young, 2020-11-04:
I tested the sample code by changing my EndNode to work.(96b_wistrio)
I have a custom LoRaWAN Gateway.

In my case, Tx Timeout did not occur.(`KR920`)
Test Commit SHA: 0d4dca10b2ac156b598f96b4dadb44ac7001380b

**EndNode**(96b_wistrio)
```
*** Booting Zephyr OS build zephyr-v2.4.0-1226-g0d4dca10b2ac  ***
[00:00:00.200,000] <inf> sx1276: SX1276 Version:12 found
[00:00:00.283,000] <inf> lorawan_class_a: Joining network over OTAA
[00:00:09.647,000] <inf> lorawan: Joined network! DevAddr: 26042e41
[00:00:09.648,000] <inf> lorawan_class_a: Sending data...
[00:00:14.158,000] <inf> lorawan_class_a: Data sent!
[00:00:28.669,000] <inf> lorawan_class_a: Data sent!
[00:00:43.178,000] <inf> lorawan_class_a: Data sent!
[00:00:57.688,000] <inf> lorawan_class_a: Data sent!
[00:01:12.198,000] <inf> lorawan_class_a: Data sent!
[00:01:26.710,000] <inf> lorawan_class_a: Data sent!
[00:01:41.220,000] <inf> lorawan_class_a: Data sent!
[00:01:55.730,000] <inf> lorawan_class_a: Data sent!
[00:02:10.242,000] <inf> lorawan_class_a: Data sent!
[00:02:24.755,000] <inf> lorawan_class_a: Data sent!
[00:02:39.264,000] <inf> lorawan_class_a: Data sent!
```

**LoRaWAN Gateway Console**
```
INFO: Received pkt from mote: D00257FF (fcnt=46037)

JSON up: {"rxpk":[{"tmst":1001228420,"chan":2,"rfch":0,"freq":922.500000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":12.5,"rssi":-17,"size":23,"data":"AP9XAtB+1bNw663OiVHkfwDyzapSCsY="}]}
INFO: [up] PUSH_ACK received in 68 ms
INFO: [down] PULL_ACK received in 68 ms
INFO: [down] PULL_RESP received  - token[0:42] :)

JSON down: {"txpk":{"imme":false,"tmst":1007228420,"freq":921.9,"rfch":0,"powe":23,"modu":"LORA","datr":"SF12BW125","codr":"4/5","ipol":true,"size":33,"ncrc":true,"data":"IIuhC3RRUmLLEU5D4PL2ScwEiNaJu5C+6CtaF6hEW/y2"}}
INFO: tx_start_delay=1495 (1495.500000) - (1497, bw_delay=1.500000, notch_delay=0.000000)
Info: packet will be sent without CRC
INFO: Enabling TX notch filter
66.79.99.3c.9.10.2d.c.0.1c.21.18.0.8.0.0.20.8b.a1.b.74.51.52.62.cb.11.4e.43.e0.f2.f6.49.cc.4.88.d6.89.bb.90.be.e8.2b.5a.17.a8.44.5b.fc.b6.end
INFO: [down] PULL_ACK received in 69 ms
lgw_receive:1156: FIFO content: 1 5d 1 5 17
lgw_receive:1175: [0 17]
Note: LoRa packet

INFO: Received pkt from mote: 260428D0 (fcnt=1)

JSON up: {"rxpk":[{"tmst":1010565500,"chan":0,"rfch":0,"freq":922.100000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":10.5,"rssi":-26,"size":23,"data":"gNAoBCYAAQACTlvbehz8OmUyK9CwItc="}]}
INFO: [up] PUSH_ACK received in 69 ms
INFO: [down] PULL_RESP received  - token[0:43] :)

JSON down: {"txpk":{"imme":false,"tmst":1012565500,"freq":921.9,"rfch":0,"powe":23,"modu":"LORA","datr":"SF12BW125","codr":"4/5","ipol":true,"size":12,"ncrc":true,"data":"YNAoBCYgAADRhLDW"}}
INFO: [down] PULL_ACK received in 68 ms
```

**TTN Network**(Gateway Console - Traffic)
[image]

vtothsvk, 2020-11-04:
Tried both [0d4dca1](https://github.com/zephyrproject-rtos/zephyr/commit/0d4dca10b2ac156b598f96b4dadb44ac7001380b) and the LoRaMac node 4.4.5. Unfortunately, to no avail. Still getting the Tx timeout.

Still using EU433 and tried both ABP and OTAA.

```
*** Booting Zephyr OS build zephyr-v2.4.0-1226-g0d4dca10b2ac  ***<\r><\n>
[00:00:00.200,000] <27>[0m<inf> sx1276: SX1276 Version:12 found<27>[0m<\r><\n>
[00:00:00.263,000] <27>[0m<inf> lorawan_class_a: Joining network over OTAA<27>[0m<\r><\n>
[00:00:04.314,000] <27>[1;31m<err> lorawan: MlmeConfirm failed : Tx timeout<27>[0m<\r><\n>
[00:00:04.314,000] <27>[1;31m<err> lorawan_class_a: lorawan_join_network failed: -116<27>[0m<\r><\n>
```

```
*** Booting Zephyr OS build zephyr-v2.4.0-1226-g0d4dca10b2ac  ***<\r><\n>
[00:00:00.200,000] <27>[0m<inf> sx1276: SX1276 Version:12 found<27>[0m<\r><\n>
[00:00:00.267,000] <27>[0m<inf> lorawan_class_a: Joining network over ABP<27>[0m<\r><\n>
[00:00:00.267,000] <27>[0m<inf> lorawan_class_a: Sending data...<27>[0m<\r><\n>
[00:00:04.282,000] <27>[1;31m<err> lorawan: McpsRequest failed : Tx timeout<27>[0m<\r><\n>
[00:00:04.282,000] <27>[1;31m<err> lorawan_class_a: lorawan_send failed: -116<27>[0m<\r><\n>
```

KwonTae-young, 2020-12-03:
@vtothsvk Looking at TTN's [Frequency Plan](https://www.thethingsnetwork.org/docs/lorawan/frequency-plans.html), it does not support EU433.     
[image]

vtothsvk, 2020-12-03:
> @vtothsvk Looking at TTN's [Frequency Plan](https://www.thethingsnetwork.org/docs/lorawan/frequency-plans.html), it does not support EU433.
> [image]

Tried ChirpStack as well. Still got the tx-timeout

KwonTae-young, 2020-12-30:
I have a problem configuring the ChirpStack environment, so I can't test it. :(

Mani-Sadhasivam, 2021-01-17:
@vtothsvk Can you confirm if the gateway works with RakWireless SDK running on Wistrio?

vtothsvk, 2021-01-17:
> @vtothsvk Can you confirm if the gateway works with RakWireless SDK running on Wistrio?

Yes it does, when using ChirpStack that is (because of the lack of EU433 support of the TTN)

carlescufi, 2021-01-26:
Lowering severity due to this affecting a single board in a single subsystem.

carlescufi, 2021-02-09:
PR updating loramac-node to v4.4.5: https://github.com/zephyrproject-rtos/loramac-node/pull/7

closed 2021-04-26 by github-actions[bot]