zephyrproject-rtos/zephyr #29589: Creating a k_thread and k_sem as runtime instantiated a kernel object causes general protection fault

maxbachmann, 2020-10-27:
**Describe the bug**
When I try to create a `k_thread` and a `k_sem` using a runtime instantiated a kernel object in Userspace (CONFIG_USERSPACE and CONFIG_DYNAMIC_OBJECTS are enabled) Zephyr throws a General protection fault. I did create a minimal reproducible example, but I can not figure out what I did wrong. The crash occurs when `k_thread_create` is called. However it does not occur unless I create a `k_sem` object using `k_object_alloc(K_OBJ_SEM)` beforehand.

I only tried this on the up quared board, since qemu is stuck while booting for some reason.

**To Reproduce**
Steps to reproduce the behavior:
1. Download [Error_Minimal.zip](https://github.com/zephyrproject-rtos/zephyr/files/5446746/Error_Minimal.zip) and unzip
1. mkdir build; cd build
2. cmake -DBOARD=up_squared ..
3. make
4. run on up squared board

**Expected behavior**
Both Thread and Semaphore should be created and the thread should print a counter message every 250ms.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.4.0-1054-g1ae2b7cad9db  ***
created semaphore
[00:00:00.084,000] <err> os: General protection fault (code 0x0)
[00:00:00.084,000] <err> os: RAX: 0x0000000000000000 RBX: 0x00000000001141f8 RCX: 0x000000000000000c RDX: 0x000000000112f000
[00:00:00.084,000] <err> os: RSI: 0x000000000112f000 RDI: 0x00000000001141f8 RBP: 0x000000000112ffe8 RSP: 0x0000000001536e98
[00:00:00.084,000] <err> os:  R8: 0x00000000001141f8  R9: 0x000000000193a000 R10: 0x000000000010f280 R11: 0x000000000000001c
[00:00:00.084,000] <err> os: R12: 0x0000000000000000 R13: 0x0000000000000000 R14: 0x0000000000100212 R15: 0x0000000000000000
[00:00:00.084,000] <err> os: RSP: 0x0000000001536e98 RFLAGS: 0x0000000000010202 CS: 0x0018 CR3: 0x000000000193b000
[00:00:00.084,000] <err> os: RIP: 0x000000000000117f
[00:00:00.084,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:00.084,000] <err> os: Current thread: 0x0000000000113a80 (unknown)
[00:00:00.181,000] <err> os: Halting system
```

**Environment (please complete the following information):**
 - OS: Linux
-  Board: Up Squared
 - Toolchain Zephyr SDK
 - Commit SHA: 1ae2b7cad9db8a429072c6ee6f1532c7944f74e4

**Additional context**
Here is the minimal example so it is not required to download the zip to check the code:
```
#include <zephyr.h>
#include <sys/printk.h>
#include <kernel.h>
#include <string.h>
#include <stdlib.h>

#define MY_STACK_SIZE 0x1000
#define MY_PRIORITY 5

K_THREAD_STACK_DEFINE(my_stack_area, MY_STACK_SIZE);

void my_entry_point(void* a, void* b, void* c) {
    int counter = 0;
	while (true) {
		printk("counter %i\n", counter);
		++counter;
		k_msleep(250);
	}
}

void main(void)
{
	k_thread_system_pool_assign(k_current_get());
    struct k_sem* my_sem = (struct k_sem*)k_object_alloc(K_OBJ_SEM);

    if(NULL == my_sem){
        printk("failed allocating semaphore object\n");
        return;
    }
    k_sem_init(my_sem, 0, CONFIG_SEM_VALUE_MAX);
    printk("created semaphore\n");

	struct k_thread* my_thread = (struct k_thread*)k_object_alloc(K_OBJ_THREAD);

    if (NULL == my_thread){
		printk("failed allocating thread object\n");
        return;
	}
    k_thread_create(my_thread, my_stack_area,
                                 K_THREAD_STACK_SIZEOF(my_stack_area),
                                 my_entry_point,
                                 NULL, NULL, NULL,
                                 MY_PRIORITY,
                                 K_INHERIT_PERMS | K_USER,
								 K_NO_WAIT);

    printk("created thread\n");
    k_sleep(K_FOREVER);
}
```

andrewboie, 2020-10-27:
Thanks for the report, will take a look.

maxbachmann, 2020-10-27:
as a related question: is there something similar to `k_object_alloc` that can be used in kernel mode to create runtime instantiated a kernel objects? `k_object_alloc` is a no op in kernel mode, so I tried to simply use:
```cpp
struct k_thread* my_thread = (struct k_thread*)k_malloc(sizeof(struct k_thread));
```
which does not work properly. So I wondered whats the correct way.

andrewboie, 2020-10-28:
> I tried to simply use:
> 
> ```c++
> struct k_thread* my_thread = (struct k_thread*)k_malloc(sizeof(struct k_thread));
> ```
> 
> which does not work properly. So I wondered whats the correct way to do this in kernel mode.

Can you give some detail on what doesn't work with this?
AFAIK there's no special alignment requirements with k_thread objects in 64-bit mode x86

maxbachmann, 2020-10-29:
> Can you give some detail on what doesn't work with this?
AFAIK there's no special alignment requirements with k_thread objects in 64-bit mode x86

I figured I would ask to make sure that this is the correct way to allocate kernel objects in kernel mode. Since it appears to be I created #29629 to show a minimal example using `k_malloc` that fails on thread creation. 
Is there an alignment requirement for k_thread objects on any other platform supported by zephyr?

andrewboie, 2020-10-29:
Now I see it. k_thread objects have to be aligned on x86_64 as well, due to the `sse` member which gets floating point context saved to it:

```
struct _thread_arch {
	uint8_t flags;

#ifdef CONFIG_USERSPACE
	/* Physical address to page tables used by this thread. Supervisor
	 * threads always use the kernel's page table, user thread use
	 * per-thread tables stored in the stack object
	 */
	uintptr_t ptables;

	/* Track available unused space in the stack object used for building
	 * thread-specific page tables.
	 */
	uint8_t *mmu_pos;

	/* Initial privilege mode stack pointer when doing a system call.
	 * Un-set for supervisor threads.
	 */
	char *psp;

	/* SS and CS selectors for this thread when restoring context */
	uint64_t ss;
	uint64_t cs;
#endif

	uint64_t rax;
	uint64_t rcx;
	uint64_t rdx;
	uint64_t rsi;
	uint64_t rdi;
	uint64_t r8;
	uint64_t r9;
	uint64_t r10;
	uint64_t r11;
	char __aligned(X86_FXSAVE_ALIGN) sse[X86_FXSAVE_SIZE];
};
```

On 32 -bit there is a similar situation with `preempFloatReg` member.

Dynamic thread objects are not well supported in Zephyr. There's an old bug open about this: https://github.com/zephyrproject-rtos/zephyr/issues/17893

> I figured I would ask to make sure that this is the correct way to allocate kernel objects in kernel mode

Most users do not use an allocator at all and instantiate the thread objects they need statically. What you are doing is unusual for a microcontroller OS.

maxbachmann, 2020-10-29:
Ah yes, then this is the issue both here and in #29629

> Most users do not use an allocator at all and instantiate the thread objects they need statically. What you are doing is unusual for a microcontroller OS.

We use the same application on multiple OS and only change an OS abstraction layer, which forces us to return void* e.g. for threads.

dcpleung, 2020-10-30:
Blocked by #17893

fzawadiak, 2021-01-13:
I also noticed that when I try to allocate object from thread without active resource pool it tries to allocate from pool 0xAAAAAAAA. Adding k_thread_system_pool_assign(k_current_get()); fixed it.

maxbachmann, 2021-01-13:
> I also noticed that when I try to allocate object from thread without active resource pool it tries to allocate from pool 0xAAAAAAAA leading to a crash. Adding k_thread_system_pool_assign(k_current_get()); fixed it. Tested with STM32H747-DISCO M7. There is a check whether pool is assigned, but it checks for NULL not 0xAAAAAAAA which seems like init memory filler?

Since this is a separate issue you should probably open a new issue on this.

PR #30762 (2021-01-13): kernel: userspace: aligned memory allocation for dynamic objects
This allows allocating dynamic kernel objects with memory alignment
requirements. The first candidate is for thread objects for x86
and x86_64 where saving/restoring SSE/FPU registers require strict
alignment.
    
Fixes #29589
Fixes #29629
commit 7a5f9e81 (2021-01-13): x86: define dynamic thread object alignment

x86 and x86_64 require certain alignment in the k_thread struct
since the buffer to save/restore FPU/SSE registers requires
strict alignment.

Fixes #29589
Fixes #29629
closed 2021-01-13 by andrewboie