zephyrproject-rtos/zephyr #29632: GPIO interrupt support for IO expander

usanskrati, 2020-10-29:
There is now GPIO interrupt support for IO expander in Zephyr. So, now required is  the GPIO overlay in GPIO test to  include a pin for  **Microchip board mec15xxevb_assy6853**.

ashaannajoy, 2020-11-12:
We need  gpio  expander interrupt callback in microchip board similar to  https://github.com/zephyrproject-rtos/zephyr/pull/28308/

ashaannajoy, 2020-11-19:
There is now GPIO interrupt support  for IO expander in Zephyr. https://github.com/zephyrproject-rtos/zephyr/pull/28308/

but not for microchip 15021, 1521. The support is for other boards

ashaannajoy, 2020-12-09:
Clarifying that this is not a test but an ask to add interrupt support for gpio IO expander chip in microchip gpio driver, similar to the implementation done for  #28308

closed 2021-02-04 by jocelyn-li

chen-png, 2021-02-04:
the GPIO expander on microchip board is the same as in #28308, this has already been enabled, and used existing gpio_basc_api to test, it was ok, so closed this