zephyrproject-rtos/zephyr #29657: enc28j60 on nRF52840 stalls during enc28j60_init_buffers in zephyr 2.4.0

qdot, 2020-10-30:
**Describe the bug**
When booting a enc28j60 on a nRF52840 with custom pin assignments for the SPI 0 bus, a copy of the DHCP client example works fine for Zephyr v2.3.0, but stalls in the enc28j60_init_buffers function on Zephyr v2.4.0, specifically in this loop: https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/ethernet/eth_enc28j60.c#L349

**To Reproduce**
Steps to reproduce the behavior:
1. Compile https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/net/dhcpv4_client for `nrf52840dk_nrf52840` the following overlay:

```
&spi0 {
	compatible = "nordic,nrf-spi";
	status = "okay";
	sck-pin = <13>;
	mosi-pin = <16>;
	miso-pin = <12>;
	cs-gpios = <&gpio0 11 0>;

	eth_enc28j60: eth_enc28j60@0 {
		compatible = "microchip,enc28j60";
		reg = <0x0>;
		local-mac-address = [00 04 a3 11 92 83];
		spi-max-frequency = <10000000>;
		int-gpios = <&gpio0 29 GPIO_ACTIVE_LOW>;
		label = "ENC28J60";
	};
};
```

2. Change to west project directory from a v2.3.0 version to a v2.4.0 version.

**Expected behavior**
DHCP example should come up fine, does so in v2.3.0

**Impact**
Showstopper, firmware no longer boots in v2.4.0. Never makes it to main() function, just sticks in confirmation loop forever.

**Logs and console output**
No log output as of yet, and not sure if this is a common setup. Haven't stepped through disassembly to see differences yet.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: Zephyr 2.3 (Works), Zephyr 2.4 (Fails), GNU ARM 09 2020 toolchain
 - Commit SHA or Version used: zephyr-v2.3.0 and zephyr-v2.4.0

anangl, 2020-11-19:
@qdot The problem is in the overlay that you use. This line:
```
cs-gpios = <&gpio0 11 0>;
```
should contain the proper GPIO flags for the CS line in the last item of the array, so you should use the following:
```
cs-gpios = <&gpio0 11 GPIO_ACTIVE_LOW>;
```
In Zephyr v2.3.0 this last item is ignored, hence the overlay works with that version. But it won't with v2.4.0.

qdot, 2020-11-19:
@anangl Thanks! I'll try this.

Just curious, is there any sort of device tree overlay validator I could check for issues like this with in the future?

qdot, 2020-11-19:
Just tested, this fix works for me in Zephyr 2.4, closing issue (but would still love guidance on debugging things like this in the future!)

closed 2020-11-19 by qdot