zephyrproject-rtos/zephyr #29675: Remove pinmux dependency on STM32 boards

Nukersson, 2020-10-31:
Removing pinmux.c of each of ST's Zephyr-Tree boards can be done now after DT approach were introduced. My new PR #29674 shows first feasible sample.

lochej, 2020-10-31:
Thank you for creating a tracking issue for that one 👍 
Shall we try to automate that process ? 

As far as I know, as long as pinmux.c pin configuration array is empty, then pinmux.c can be safely removed.

Only I remember of a board nucleo_g0 (the only 8 pin one from ST) that had a particular use of pinmux for some remap functionnality.

Nukersson, 2020-10-31:
> Thank you for creating a tracking issue for that one 👍
> Shall we try to automate that process ?

It is definitely a good idea. But I would like first the guys from ST take a look at #29674. Maybe they have something to point.

> As far as I know, as long as pinmux.c pin configuration array is empty, then pinmux.c can be safely removed.

I've tested it in my PR and can confirm it.
 
> Only I remember of a board STM32G0316-DISCO (the only 8 pin one from ST) that has a particular use of pinmux for some remap functionnality.

Very good, that you noticing it.

erwango, 2020-11-02:
@Nukersson I plan to do all at once after the defines are deprecated. Working on it.

Nukersson, 2020-11-02:
> @Nukersson I plan to do all at once after the defines are deprecated. Working on it.

Thanks. I've looked in issues first and found nothing. Can you at least add an issue first, if you planning to do something obvious? It could then help us together to consolidate the ressources. Anyway, with this approach my board can perfectly live without pinmux.c

erwango, 2020-11-04:
Fixed byhttps://github.com/zephyrproject-rtos/zephyr/pull/29723

closed 2020-11-04 by erwango