zephyrproject-rtos/zephyr #29683: BLE IPSP sample doesn't work on raspberry pi 4 with nrf52840_mdk board

hguomin8, 2020-11-01:
**Describe the bug**
Follow the steps in the IPSP sample document to test BLE IPSP connection between RPI4 and nrf52840_mdk board, while I can ping nrf52840_mdk successfully with scoped IPv6 address, but can't ping with static ipv6 address, also can't connect to the board using telnet, please check the screen shot below.

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr\samples\bluetooth\ipsp
2. west build -b nrf52840_mdk . -DCONF_FILE="prj_dbg.conf"
3. west flash
4. See error in below screen shot

**Expected behavior**
Can ping and telnet via static ipv6 address

**Impact**
Can not make any progress without this ipsp feature...

**Logs and console output**
1. Steps and ouputs on RPI 4
[image]

2. net shell logs on nrf52840_board
[image]

**Environment (please complete the following information):**
 - OS: (Raspberry pi OS(buster, kernel 5.4.51-v7l+))

**Additional context**
NA

Vudentz, 2020-11-05:
We will probably need some HCI logs, but it does looks like this is a Linux problem rather than Zephyr.

hguomin8, 2020-11-06:
Hmm...this issue can be reproduced easily, look forward you guys can help to fix it, I am not specialize in kernel development

carlescufi, 2020-11-12:
@hguomin8 could you please report this to the [Linux BlueZ mailing list](http://www.bluez.org/development/lists/), since this looks like a BlueZ issue? I will close this issue but feel free to reopen it if you find out this to be a Zephyr issue instead.

closed 2020-11-12 by carlescufi