zephyrproject-rtos/zephyr #29684: Can not make multiple BLE IPSP connection to the same host

hguomin8, 2020-11-01:
**Describe the bug**
I am using a RPI 3B+ running Raspberry OS(Kernel 4.19.97) as gateway, and two nrf52840_mdk board as ble ipsp nodes running Zephyr v2.3.0, each of above devices' static ipv6 address is as below:
- RPI 3B+: 2001:db8::2
- ipsp node1: 2001:db8::1
- ipsp node2: 2001:db8::3 

[the IPSP sample code](https://docs.zephyrproject.org/latest/samples/bluetooth/ipsp/README.html) is used as ipsp nodes program.

the issue is that, when I only connect  any of 1 ipsp node with RPI, the ipsp connection work fine, then after connect the other one ipsp node to RPI, both of connection between ipsp node and RPI will lost. 

My questions is that, why the ble ipsp can only make 1:1 connection? but not works in 1:n connection scenarios? 

Thanks.

hguomin8, 2020-11-05:
@Vudentz any comments for the issue? or any walk around to make this 1:n connection works? Thanks!

Vudentz, 2020-11-05:
Well the nodes still only have 1 connection so Im afraid your problem is actually on the Linux side which acts as the router, if I recall correctly there was a problem with the address resolution on the Linux side but I though we had it fixed some time ago but perhaps not in the kernel you are using.

hguomin8, 2020-11-06:
> Well the nodes still only have 1 connection so Im afraid your problem is actually on the Linux side which acts as the router, if I recall correctly there was a problem with the address resolution on [...]

Thanks for this info, so may I know which version of the kernel fixed this issue? or can I have that patch to make it works on current kernel(4.19.97)? I tried to upgrade to latest RPI OS which has newer kernel but it will have the #29683 issue...

carlescufi, 2020-11-12:
@hguomin8 could you please report this to the [Linux BlueZ mailing list](http://www.bluez.org/development/lists/), since this looks like a BlueZ issue? I will close this issue but feel free to reopen it if you find out this to be a Zephyr issue instead.

closed 2020-11-12 by carlescufi

hguomin8, 2020-11-13:
@vcgomes I am not sure where the issue would be, because I am not a expert in linux kernel side, so could you please help to forward this issue to them and check if they can help or not? much thanks.

hguomin8, 2020-11-15:
@carlescufi I do more tests with "tcpdump -i bt0" for this issue at the weekend, it looks like the issue come from zephyr but not Bluez, please check below.

IPv6 address for each devices:
> RPI 3B+: 2001:db8::2
> ipsp node1: global addr: 2001:db8::3, link-local addr: fe80::f2f0:1ff:fec6:c12e
> ipsp node2: global addr: 2001:db8::1, link-local addr: fe80::c387:ecff:fee8:e79a

Connect node1 with 'echo "connect F2:F0:01:C6:C1:2E 2" > /sys/kernel/debug/bluetooth/6lowpan_control', ping from node1 to rpi works fine:
[image]

and ping from rpi to node1 also works fine:
[image]

Keep the ping from rpi to node1 running, then connect node2 with 'echo "connect C3:87:EC:E8:E7:9A 2" > /sys/kernel/debug/bluetooth/6lowpan_control', at this time rpi is keep sending ping but didn't get any reply from node1:
[image]
try to ping from node1 to rpi, we saw that node1 send NS to query who has 2001:db8::2(rpi) and rpi replied NA to node1, but when rpi send NS to query who is 2001:db8::3(node1), rpi didn't get any reply from node1...
[image]

at this point ping from rpi to node1 or node2,  rpi is keep sending ping message, but didn't get any reply from node1/node2
[image]

Although can't ping each other with global ipv6 address, but it's ok to ping with the link local address:
[image]
[image]
[image]

So it looks like the some issues from zephyr, could you please help to check it again? thanks.