zephyrproject-rtos/zephyr #29689: tests: drivers: gpio: gpio_basic_api: correct dts binding

cfriedt, 2020-11-01:
**Is your enhancement proposal related to a problem? Please describe.**
As @sjg20 [pointed out](https://github.com/zephyrproject-rtos/zephyr/pull/26484#discussion_r469589615) in #26484, compatible strings should use '-' instead of '_'. 

**Describe the solution you'd like**
The existing "test,gpio_basic_api" binding should be changed to "test,gpio-basic-api". All of the boards / relevant files in `tests/drivers/gpio/gpio_basic_api` should be corrected to reflect that change.

**Describe alternatives you've considered**

**Additional context**

pabigot, 2020-11-02:
While it's true that the vast majority of `compatible` strings use `-` as separators the devicetree specification does not mandate or even mention this, and there is existing practice in Zephyr and in Linux for using `_` as a separator, particularly but not exclusively when there is no `vendor,` prefix.

I don't object to making this change if we agree and document that's a standard we should follow, and add checks to the tooling to warn when people violate it.  @sjg20 @mbolivar @galak

pabigot, 2020-11-02:
BTW: I believe the reason I went with `test,gpio_basic_api` originally is because this is the binding used for the `gpio_basic_api` test (not the `gpio-basic-api` test).  I'm still a little bit inclined to that as an exception, especially since it's essentially a private binding.

galak, 2020-11-02:
> I don't object to making this change if we agree and document that's a standard we should follow, and add checks to the tooling to warn when people violate it. @sjg20 @mbolivar @galak

While it seems general convention to use `-` instead of `_` there isn't anything in the DTS specs around conventions for `compatible` beyond `vendor,` prefix usage.  So if someone wants to propose that as a change to the DTS spec than I'm all for adding a warning in the tooling if it is accepted (https://github.com/devicetree-org/devicetree-specification).

sjg20, 2020-11-04:
Certainly this is a pretty strong convention. The DT spec examples all use hyphen and the the /aliases node requires it, but oddly it has capital letters for the alias labels.

This convention is a bit confusing in Zephyr because hyphen gets converted to underscore to become a valid C identifier

The underscore is reserved for phandle names.

sjg20, 2020-11-04:
@cfriedt I'm happy to have a crack at a DT spec update if you don't want to do it.

cfriedt, 2020-11-04:
> @cfriedt I'm happy to have a crack at a DT spec update if you don't want to do it.

I think it's unrelated to this ticket for the most part, so please go ahead.

sjg20, 2021-04-01:
Applied here:

closed 2021-04-01 by sjg20