zephyrproject-rtos/zephyr #29696: [Coverity CID :215241] Buffer not null terminated in drivers/modem/hl7800.c

zephyrbot, 2020-11-02:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/c9a2a5e7fb0194bfb168f5aa1a0a83c73f62acb3/drivers/modem/hl7800.c

Category: Memory - illegal accesses
Function: `send_at_cmd`
Component: Drivers
CID: [215241](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=215241)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

rerickson1, 2021-01-21:
Can I get some details on what the issue is here?

carlescufi, 2021-01-26:
> Can I get some details on what the issue is here?

```

 854/* Send an AT command with a series of response handlers */
 855static int send_at_cmd(struct hl7800_socket *sock, const uint8_t *data,
 856                       k_timeout_t timeout, int retries, bool no_id_resp)
 857{
 858        int ret = 0;
 859
 860        ictx.last_error = 0;
 861
 862        do {
    	1. Condition !sock, taking true branch.
 863                if (!sock) {
 864                        k_sem_reset(&ictx.response_sem);
 865                        ictx.last_socket_id = 0;
    	2. Falling through to end of if statement.
 866                } else {
 867                        k_sem_reset(&sock->sock_send_sem);
 868                        ictx.last_socket_id = sock->socket_id;
 869                }
    	3. Condition no_id_resp, taking true branch.
 870                if (no_id_resp) {
    	
CID 215241 (#1 of 1): Buffer not null terminated (BUFFER_SIZE_WARNING)
4. buffer_size_warning: Calling strncpy with a maximum size argument of 32 bytes on destination array ictx.no_id_resp_cmd of size 32 bytes might leave the destination string unterminated.
 871                        strncpy(ictx.no_id_resp_cmd, data,
 872                                sizeof(ictx.no_id_resp_cmd));
 873                        ictx.search_no_id_resp = true;
 874                }
 875
    	5. Condition 0 /* 1 && (4U <= 0 || ((0 == 0 && 4U <= __log_level) && 4U <= 4)) */, taking false branch.
    	6. Condition 0, taking false branch.
 876                LOG_DBG("OUT: [%s]", log_strdup(data));
 877                mdm_receiver_send(&ictx.mdm_ctx, data, strlen(data));
 878                mdm_receiver_send(&ictx.mdm_ctx, "\r", 1);
 879
    	7. Condition timeout.ticks == k_timeout_t({}).ticks, taking true branch.
 880                if (K_TIMEOUT_EQ(timeout, K_NO_WAIT)) {
    	8. Jumping to label done.
 881                        goto done;
 882                }
 883
 884                if (!sock) {
 885                        ret = k_sem_take(&ictx.response_sem, timeout);
 886                } else {
 887                        ret = k_sem_take(&sock->sock_send_sem, timeout);
 888                }
 889
 890                if (ret == 0) {
 891                        ret = ictx.last_error;
 892                } else if (ret == -EAGAIN) {
 893                        ret = -ETIMEDOUT;
 894                }
 895
 896                retries--;
 897                if (retries < 0) {
 898                        retries = 0;
 899                }
 900        } while (ret != 0 && retries > 0);
```

carlescufi, 2021-01-26:
[image]

closed 2021-01-27 by nashif