zephyrproject-rtos/zephyr #29702: [Coverity CID :215232] Dereference after null check in subsys/emul/emul_bmi160.c

zephyrbot, 2020-11-02:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/c9a2a5e7fb0194bfb168f5aa1a0a83c73f62acb3/subsys/emul/emul_bmi160.c

Category: Null pointer dereferences
Function: `bmi160_emul_io_spi`
Component: Other
CID: [215232](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=215232)

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

simonguinot, 2020-11-02:
Hi @nashif, I suspect you got the wrong Simon :)

sjg20, 2020-11-04:
https://github.com/zephyrproject-rtos/zephyr/pull/29799

PR #29799 (2020-11-19): emul: spi: bmi160: Add a check for a missing rx buffer
If a read is requested without a read buffer, the emulator currently
crashes. Fix this by adding a check.

Fixes: #29703
Fixes: #29702
Fixes: #29017
Fixes: #29016
commit 129cfec2 (2020-11-19): emul: spi: bmi160: Add a check for a missing rx buffer

If a read is requested without a read buffer, the emulator currently
crashes. Fix this by adding a check.

Fixes: #29703
Fixes: #29702
Fixes: #29017
Fixes: #29016
closed 2020-11-19 by MaureenHelm