zephyrproject-rtos/zephyr #29705: reel_board:tests/kernel/sched/schedule_api/ failed

s-kelvin, 2020-11-02:
**Describe the bug**
Running tests/kernel/sched/schedule_api/ on reel board. Assertion in test_slice_reset failed in both kernel.scheduler and kernel.scheduler.multiq testcases. 

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b reel_board tests/kernel/sched/schedule_api/
2. west flash

**Expected behavior**
Test should pass.

**Impact**
Test suite fails.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.4.0-1133-ge4e3ab3cc315  ***
Running test suite threads_scheduling
===================================================================
START - test_bad_priorities
PASS - test_bad_priorities
===================================================================
START - test_priority_cooperative
PASS - test_priority_cooperative
===================================================================
START - test_priority_preemptible
PASS - test_priority_preemptible
===================================================================
START - test_priority_preemptible_wait_prio
PASS - test_priority_preemptible_wait_prio
===================================================================
START - test_yield_cooperative
PASS - test_yield_cooperative
===================================================================
START - test_sleep_cooperative
PASS - test_sleep_cooperative
===================================================================
START - test_sleep_wakeup_preemptible
PASS - test_sleep_wakeup_preemptible
===================================================================
START - test_pending_thread_wakeup
PASS - test_pending_thread_wakeup
===================================================================
START - test_time_slicing_preemptible
PASS - test_time_slicing_preemptible
===================================================================
START - test_time_slicing_disable_preemptible
PASS - test_time_slicing_disable_preemptible
===================================================================
START - test_lock_preemptible
PASS - test_lock_preemptible
===================================================================
START - test_unlock_preemptible
PASS - test_unlock_preemptible
===================================================================
START - test_unlock_nested_sched_lock
PASS - test_unlock_nested_sched_lock
===================================================================
START - test_sched_is_preempt_thread
PASS - test_sched_is_preempt_thread
===================================================================
START - test_slice_reset
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:91: thread_time_slice: (t <= expected_slice_max is false)
timeslice too big, expected 6555 got 7208
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sched/schedule_api/src/test_sched_timeslice_reset.c:88: thread_time_slice: (t >= expected_slice_min is false)
timeslice too small, expected 6553 got 617
FAIL - test_slice_reset
===================================================================
START - test_slice_scheduling
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
PASS - test_slice_scheduling
===================================================================
START - test_priority_scheduling
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
ABCDEFGHI
PASS - test_priority_scheduling
===================================================================
START - test_wakeup_expired_timer_thread
PASS - test_wakeup_expired_timer_thread
===================================================================
START - test_user_k_wakeup
PASS - test_user_k_wakeup
===================================================================
START - test_user_k_is_preempt
PASS - test_user_k_is_preempt
===================================================================
Test suite threads_scheduling failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 18.04.4 LTS
 - Toolchain Zephyr SDK 0.11.2

maksimmasalski, 2020-11-02:
Fails on the mec15xxevb_assy6853 too. In multiq configuration and in standard without it.

pabigot, 2020-11-03:
This is specific to `CONFIG_TIMESLICING=y`, was introduced at 0875740a3457ecfddc70c93cacd44932f64058a4, and is reproducible on multiple Nordic boards.

Given that the previous sync waited for a "millisecond" boundary I'm wondering if the test has an implicit assumption that milliseconds can be precisely represented, which is not true on Nordic hardware.

andyross, 2020-11-17:
Was just pointed to this.  Indeed, this test looks to me like it's doing ms math on its timeouts.

maksimmasalski, 2020-11-18:
> Was just pointed to this. Indeed, this test looks to me like it's doing ms math on its timeouts.

What time you can expect to submit a PR? Do you need any help?

ABOSTM, 2021-01-15:
On STM32 boards, I found that this test fails depending on clock frequency,
and I propose a fix by increasing SLICE_SIZE:
https://github.com/zephyrproject-rtos/zephyr/pull/30791

PR #30916 (2021-01-27): sched: timeout: Do not miss slice timeouts
Time slices don't have a timeout struct associated and stored in
timeout_list. Time slice timeout is direct programmed in the system
clock and tracked in _current_cpu->slice_ticks.

There is one issue where the time slice timeout can be missed because
the system clock is re-programmed to a longer timeout. To this happens,
it is only necessary that the timeout_list is empty (any timeout set)
and a new timeout longer than remaining time slice is set. This is cause
because z_add_timeout does not check for the slice ticks.

Fixes #29705
commit 2f6b202e (2021-01-27): tests/schedule_api: Correct tick/ms math for non-divisible clocks

Adjusting the tick alignment of this test caused it to start breaking
on nRF5 platforms, which use a 32768 Hz clock that doesn't divide
evenly into the millisecond precision used by the test.  The "half
slice" math ended up being wrong by a bit.

Convert to ticks first before computing the cycle delay needed.

Fixes #29705
closed 2021-01-27 by nashif