zephyrproject-rtos/zephyr #29710: Building USB Mass sample hal_nxp failed

jturnsek, 2020-11-02:
Latest changes to hal_nxp mcux USB middleware trigger build time error.

jfischer-phytec-iot, 2020-11-02:
By commit 8b2406d62e1a9ca44d52bf45e0e3a84c1e741f63 from https://github.com/zephyrproject-rtos/hal_nxp/pull/56

galak, 2020-11-02:
Seeing failure on USB tests for sanitycheck on i.mx-rt platforms:

```
./scripts/sanitycheck -s tests/subsys/usb/device/usb.device  -l

ERROR   - mimxrt1060_evk_hyperflash tests/subsys/usb/device/usb.device                 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/mimxrt1060_evk_hyperflash/tests/subsys/usb/device/usb.device/build.log
INFO    - Total complete:    1/  65   1%  skipped:    0, failed:    1
ERROR   - mimxrt1064_evk            tests/subsys/usb/device/usb.device                 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/mimxrt1064_evk/tests/subsys/usb/device/usb.device/build.log
INFO    - Total complete:    2/  65   3%  skipped:    0, failed:    2
ERROR   - mimxrt1060_evk            tests/subsys/usb/device/usb.device                 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/mimxrt1060_evk/tests/subsys/usb/device/usb.device/build.log
INFO    - Total complete:   22/  65  33%  skipped:    0, failed:    3
ERROR   - mimxrt1015_evk            tests/subsys/usb/device/usb.device                 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/mimxrt1015_evk/tests/subsys/usb/device/usb.device/build.log
INFO    - Total complete:   23/  65  35%  skipped:    0, failed:    4
ERROR   - mimxrt1010_evk            tests/subsys/usb/device/usb.device                 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/mimxrt1010_evk/tests/subsys/usb/device/usb.device/build.log
INFO    - Total complete:   47/  65  72%  skipped:    0, failed:    5
ERROR   - mimxrt1050_evk_qspi       tests/subsys/usb/device/usb.device                 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/mimxrt1050_evk_qspi/tests/subsys/usb/device/usb.device/build.log
INFO    - Total complete:   49/  65  75%  skipped:    0, failed:    6
ERROR   - mimxrt1050_evk            tests/subsys/usb/device/usb.device                 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/mimxrt1050_evk/tests/subsys/usb/device/usb.device/build.log
INFO    - Total complete:   54/  65  83%  skipped:    0, failed:    7
ERROR   - mimxrt1020_evk            tests/subsys/usb/device/usb.device                 FAILED: Build failure
ERROR   - see: /home/galak/git/zephyr/sanity-out/mimxrt1020_evk/tests/subsys/usb/device/usb.device/build.log
INFO    - Total complete:   65/  65  100%  skipped:    0, failed:    8
```

PR #29797 (2020-11-06): west.yml: Pull in build fix for NXP USB driver
The SDK USB driver was missing some defines that were accidentally deleted when updating to SDK 2.8

Depends on zephyrproject-rtos/hal_nxp#66
Fixes #29710
closed 2020-11-06 by galak