zephyrproject-rtos/zephyr #29711: Add BSD socket option SO_RCVTIMEO

Delsian, 2020-11-02:
Current implementation recvfrom() has no possibility to exit after the timeout, but Linux socket has option SO_RCVTIMEO

https://linux.die.net/man/7/socket

Usage:
```
struct timeval tv;
tv.tv_sec = 0;
tv.tv_usec = 100000;
if (setsockopt(rcv_sock, SOL_SOCKET, SO_RCVTIMEO,&tv,sizeof(tv)) < 0) {
    perror("Error");
}
if(recvfrom(rcv_sock, ackBuffer,sizeof(ackBuffer), 0,
       (struct sockaddr *) &servAddr2, &fromSize) < 0){
    //timeout reached
    ...
}
```

jukkar, 2020-11-02:
I have no objections to this feature. Are you planning to implement this or do you wish that someone else would do the work?

Delsian, 2020-11-03:
Actually I have no plans to implement this, but I strictly need any solution to correctly shut down the receiving socket after timeout.

jukkar, 2021-01-20:
The SO_RCVTIMEO options was just implemented by #31369 so closing this one.

closed 2021-01-20 by jukkar