zephyrproject-rtos/zephyr #29713: nRF5340 - duplicate unit-address

koffes, 2020-11-02:
**Describe the bug**
During building, the following warnings are printed:

```
nrf5340dk_nrf5340_cpuapp.dts.pre.tmp:54.42-66.3: Warning (unique_unit_address_if_enabled): /soc/peripheral@50000000/flash-controller@39000: duplicate unit-address (also used in node /soc/peripheral@50000000/kmu@39000)

nrf5340dk_nrf5340_cpuapp.dts.pre.tmp:328.19-334.3: Warning (unique_unit_address_if_enabled): /soc/peripheral@50000000/clock@5000: duplicate unit-address (also used in node /soc/peripheral@50000000/power@5000)

nrf5340dk_nrf5340_cpuapp.dts.pre.tmp:584.31-585.5: Warning (unique_unit_address_if_enabled): /reserved-memory/image@20000000: duplicate unit-address (also used in node /reserved-memory/image_s@20000000)
```

**To Reproduce**
Compile any sample such as `samples/basic/button` using `west build -b nrf5340dk_nrf5340_cpuapp`

**Environment (please complete the following information):**
 - Linux
 - gcc-arm-none-eabi-9-2019-q4-major
 - b3ff89bd511afef484c193a428633f6a1c866944

koffes, 2021-01-04:
This is still an issue on master today. @carlescufi : Is this an issue which should be prioritised? Having warnings in pure vanilla samples is highly undesirable.

ioannisg, 2021-01-04:
It is prioritized correctly, even LOW priority bugs count towards release criteria.
The problem is that this is not a trivial fix, considering that the peripheral base addresses _are_ identical in these nRF peripherals, so there's something to be done with DTS...

Perhaps @anangl have an idea?

carlescufi, 2021-02-03:
Adding @mbolivar-nordic  since this probably requires extending our current DT tooling to support this particular case.

mbolivar-nordic, 2021-02-03:
> Adding @mbolivar-nordic since this probably requires extending our current DT tooling to support this particular case.

We can fix this with existing tooling; it will just require copy/pasting a file around in all the nRF5340 board directories. See #31950.

PR #31954 (2021-02-04): Fix devicetree warnings
Fixes #31947 and Fixes #29713
closed 2021-02-04 by nashif