zephyrproject-rtos/zephyr #29722: West flash is not able to flash with openocd

Sergi22, 2020-11-02:
**Describe the bug**
west flash not able to find openocd. See logs.

**After cmd west flash -H:**
build configuration:
  build directory: C:\Work\zephyrproject\zephyr\build
  board: stm32f072b_disco
  runners.yaml: C:\Work\zephyrproject\zephyr\build\zephyr\runners.yaml

available runners in runners.yaml:
  openocd, jlink
default runner in runners.yaml:
  openocd
common runner configuration:
  - build_dir: C:\Work\zephyrproject\zephyr\build
  - board_dir: C:/Work/zephyrproject/zephyr/boards/arm/stm32f072b_disco
  - elf_file: C:\Work\zephyrproject\zephyr\build\zephyr\zephyr.elf
  - hex_file: C:\Work\zephyrproject\zephyr\build\zephyr\zephyr.hex
  - bin_file: C:\Work\zephyrproject\zephyr\build\zephyr\zephyr.bin
  - gdb: C:/gnuarmemb/bin/arm-none-eabi-gdb.exe
  - openocd: None
  - openocd_search: None

**To Reproduce**
Steps to reproduce the behavior:
1. Follow initial setup here - https://docs.zephyrproject.org/latest/getting_started/index.html
2. install openocd - https://docs.zephyrproject.org/latest/guides/debugging/host-tools.html#openocd-debug-host-tools
3. use west flass. See error log output in console log

**Logs and console output**

C:\Work\zephyrproject\zephyr>west flash
-- west flash: rebuilding
[0/1] cmd.exe /C "cd /D C:\Work\zephyrproject\zephyr\build...\flash && "C:\Program Files\CMake\bin\cmake.exe" -E echo "

-- west flash: using runner openocd
Traceback (most recent call last):
  File "c:\python39\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "c:\python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\User\AppData\Roaming\Python\Python39\Scripts\west.exe\__main__.py", line 7, in <module>
  File "C:\Users\User\AppData\Roaming\Python\Python39\site-packages\west\app\main.py", line 780, in main
    app.run(argv or sys.argv[1:])
  File "C:\Users\User\AppData\Roaming\Python\Python39\site-packages\west\app\main.py", line 106, in run
    self.run_command(argv)
  File "C:\Users\User\AppData\Roaming\Python\Python39\site-packages\west\app\main.py", line 339, in run_command
    self.run_extension(args.command, argv)
  File "C:\Users\User\AppData\Roaming\Python\Python39\site-packages\west\app\main.py", line 409, in run_extension
    command.run(args, unknown, self.topdir, manifest=self.manifest)
  File "C:\Users\User\AppData\Roaming\Python\Python39\site-packages\west\commands.py", line 129, in run
    self.do_run(args, unknown)
  File "C:\Work\zephyrproject\zephyr\scripts\west_commands\flash.py", line 29, in do_run
    do_run_common(self, my_args, runner_args)
  File "C:\Work\zephyrproject\zephyr\scripts/west_commands\run_common.py", line 210, in do_run_common
    runner.run(command_name)
  File "C:\Work\zephyrproject\zephyr\scripts/west_commands\runners\core.py", line 482, in run
    self.do_run(command, **kwargs)
  File "C:\Work\zephyrproject\zephyr\scripts/west_commands\runners\openocd.py", line 115, in do_run
    self.require(self.openocd_cmd[0])
  File "C:\Work\zephyrproject\zephyr\scripts/west_commands\runners\core.py", line 502, in require
    if shutil.which(program) is None:
  File "c:\python39\lib\shutil.py", line 1380, in which
    if os.path.dirname(cmd):
  File "c:\python39\lib\ntpath.py", line 223, in dirname
    return split(p)[0]
  File "c:\python39\lib\ntpath.py", line 185, in split
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain GNU_ARM_EMBEDDED

erwango, 2020-11-04:
@Sergi22 Seems that west complains on a path issue.
Can you provide the list of the west commands you run ?
Additionally, you could run commands in verbose mode for more information on your setup (`west -v flash ...`)

ABOSTM, 2020-11-05:
Hi @Sergi22,
West flash is working properly on my side.
As said @erwango, it looks like a path issue, so can you make sure you added openOCD binary directory to PATH environment variable (There is a mistake in the documentation it must be added to 'PATH' and not 'Path'):

>  On Linux, openocd is available though the Zephyr SDK. Windows users should use the following steps to install openocd:
>  * Download openocd for Windows from here: OpenOCD Windows
>  * Copy bin and share dirs to C:\Program Files\OpenOCD\
>  * Add C:\Program Files\OpenOCD\bin to ‘PATH’

Please run this command `echo $PATH` and verify it contains the openOCB directory.

Sergi22, 2020-11-05:
Problem solved, C:\Program Files\OpenOCD\bin was recognized as a PATH variable only after Windows reset. Currently, no problems with flashing.
Please, close this ticket.

closed 2020-11-05 by ABOSTM

glenn-andrews, 2021-03-08:
Getting the same problem. 

I put C:\Program Files\OpenOCD\bin in my path, and nothing happened, so I added  C:\Program Files\OpenOCD\share and C:\Program Files\OpenOCD just in case, and still no luck.

echo $path  just replies with $path but echo %path% shows all three directories in my PATH variable.

If I type openocd --version into cmd, I get the version information, so I know the path is correct.

I also tried cmd as regular user and admin, and no difference.

Any ideas?

ABOSTM, 2021-03-08:
Hi @glenn-andrews ,
This issue is already closed thanks to proper definition of PATH environment variable.
It seems your PATH variable is correct, so it is probably a different issue ...
so it is better you open a new issue with your own description, your own setup and your own error log ...

glenn-andrews, 2021-03-08:
My apologies. I'll do so now

mahammadarif, 2024-02-19:
(.venv) mdarif@mdarif:~/zephyrproject/zephyr/build$ west flash
-- west flash: rebuilding
[1/1] cd /home/mdarif/zephyrproject/zephyr/build/ze...mdarif/zephyrproject/zephyr/build/zephyr/zephyr.elf
-- west flash: using runner nrfjprog
-- runners.nrfjprog: reset after flashing requested
FATAL ERROR: Unable to find a board; is the board connected?