zephyrproject-rtos/zephyr #29732: issue with ST Nucleo H743ZI2

vagotec, 2020-11-02:
**Describe the bug**
Hi,
there is bug with ST Nucleo H743ZI2.
- build is ok
- run on Hardware not ok

[  1%] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[  2%] Built target parse_syscalls_target
[  2%] Built target kobj_types_h_target
[  2%] Built target syscall_list_h_target
[  2%] Built target driver_validation_h_target
[  2%] Built target offsets
[  3%] Built target offsets_h
[  3%] Built target zephyr_generated_headers
[ 21%] Built target kernel
[ 22%] Built target app
[ 45%] Built target zephyr
[ 46%] Built target linker_script_target
[ 48%] Built target isr_tables
[ 49%] Built target arch__common
[ 57%] Built target arch__arm__core__aarch32
[ 64%] Built target arch__arm__core__aarch32__cortex_m
[ 67%] Built target arch__arm__core__aarch32__cortex_m__mpu
[ 79%] Built target lib__libc__minimal
[ 81%] Built target lib__posix
[ 83%] Built target soc__arm__common__cortex_m
[ 84%] Built target boards__arm__nucleo_h743zi
[ 86%] Built target drivers__gpio
[ 87%] Built target drivers__serial
[ 92%] Built target ..__modules__hal__stm32__stm32cube
[ 94%] Built target zephyr_prebuilt
[ 95%] Built target linker_pass_final_script_target
[ 98%] Built target zephyr_final
[100%] Flashing nucleo_h743zi
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/sarvg/code/eclipse_cpp_ws/NUCLEO-H43ZI2/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.10.0+dev-01341-g580d06d9d-dirty (2020-06-25-12:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 1800 kHz
Info : STLINK V3J6M2 (API v3) VID:PID 0483:374E
Info : Target voltage: 3.274332
Error: jtag status contains invalid mode value - communication failure
Polling target stm32h7x.cpu0 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 100ms
Info : Previous state query failed, trying to reconnect
Error: jtag status contains invalid mode value - communication failure
Polling target stm32h7x.cpu0 failed, trying to reexamine
Examination failed, GDB will be halted. Polling again in 300ms
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32h7x.cpu0      hla_target little stm32h7x.cpu       unknown

Error: mem2array: Read @ 0x5c001004, w=4, cnt=1, failed
Error executing event examine-end on target stm32h7x.cpu0:
/home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg:199: Error: 
in procedure 'ocd_process_reset' 
in procedure 'ocd_process_reset_inner' called at file "embedded:startup.tcl", line 230
in procedure 'stm32h7x_dbgmcu_mmw' called at file "/home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 149
in procedure 'stm32h7x_mmw' called at file "/home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 224
in procedure 'stm32h7x_mrw' called at file "/home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 205
at file "/home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts/target/stm32h7x.cfg", line 199

FATAL ERROR: command exited with status 1: /home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/sarvg/zephyrproject/zephyr/boards/arm/nucleo_h743zi/support -s /home/sarvg/zephyr-sdk-0.11.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/sarvg/zephyrproject/zephyr/boards/arm/nucleo_h743zi/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase /home/sarvg/code/eclipse_cpp_ws/NUCLEO-H43ZI2/build/zephyr/zephyr.hex' -c 'reset halt' -c 'verify_image /home/sarvg/code/eclipse_cpp_ws/NUCLEO-H43ZI2/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
make[3]: *** [zephyr/cmake/flash/CMakeFiles/flash.dir/build.make:58: zephyr/cmake/flash/CMakeFiles/flash] Fehler 1
make[2]: *** [CMakeFiles/Makefile2:3341: zephyr/cmake/flash/CMakeFiles/flash.dir/all] Fehler 2
make[1]: *** [CMakeFiles/Makefile2:3348: zephyr/cmake/flash/CMakeFiles/flash.dir/rule] Fehler 2
make: *** [Makefile:729: flash] Fehler (error) 2

**Environment (please complete the following information):**
 - OS: Linux, ubuntu 20.04
 - Toolchain (Zephyr SDK: 11.0.4) and Eclipse Plug-In (Ver.: Version: 0.2.99.202009250453)

**Additional context**
Add any other context about the problem here.

gmarull, 2020-11-02:
You can try if changing reset flags like in https://github.com/zephyrproject-rtos/zephyr/pull/29714 solve the problem.

vagotec, 2020-11-03:
Thanks for this info. Sorry i am a beginer. Could you please tell me where (file or ....) i change or append that?

Thank you

gmarull, 2020-11-03:
@vagotec support/openocd.cfg file inside board folder.

vagotec, 2020-11-03:
Thanksssssssssss
my problem is solved :-)

gmarull, 2020-11-03:
@erwango we may need to update OpenOCD reset config on V3 boards.

commit 348996fb (2020-12-03): boards: nucleo_h743zi: Fix openocd configuration

Similar to what was done for other h7 based boards,
use 'connect_assert_srst' and fix init routine.

Fixes #29732
closed 2020-12-03 by nashif