zephyrproject-rtos/zephyr #29733: SAM0 will wake up with interrupted execution after deep sleep

alfonsotames, 2020-11-02:
Trying to wake up the SAM0 after a deep sleep will froze execution. The interrupt will still work but any execution will advance step by step as the interrupt is invoked. I am using an ATSAMD21G18A board.

**To Reproduce**

1. Clone the [deep sleep demo project](https://github.com/alfonsotames/deepsleep)
2. Configure LED0 and SW0 in your SAM0 dts board file.
3. Build & flash

**Expected behavior**
Pressing SW0 should should wake up SAM0 and blink LED0 5 times, then go to deep sleep again.

**Impact**
SAM0 cannot properly wake up from deep sleep.

**Environment:**
 - OS: zephyr-v2.4.0-1111-g8941f9a00c0b
- Output log to console

MaureenHelm, 2020-11-03:
@nandojve can you take a look at this?

closed 2021-01-18 by github-actions[bot]

sslupsky, 2021-05-14:
@alfonsotames Yeah, this issue is because the tickless timer is broken for sam0.  I fixed it a while ago and posted a PR for rtc_timer.c here but after two subsequent releases I gave up trying to get it into the tree because it was ignored.  Are you still using sam0?  There are other issues as well with the clocks, i2c and the uart drivers.

alfonsotames, 2021-05-14:
Thanks for your comment! No, I stopped using SAM0 since it's pointless without a proper wake up. I switched to STM32L432KC and everything looks good so far.

sslupsky, 2021-05-17:
It is possible to do (I have done so) but for what ever reason the zephyr team has not addressed the underlying issues.

nandojve, 2021-05-17:
Hi @sslupsky ,

> It is possible to do (I have done so) but for what ever reason the zephyr team has not addressed the underlying issues.

I think SAM0 doesn't have an active codeowner and that make things slow.

If you have this ready, please, rebase and resent. Link you PR with this Issue and let fix it.