zephyrproject-rtos/zephyr #29737: up_squared: tests/subsys/power/power_mgmt failed.

chen-png, 2020-11-03:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto tests/subsys/power/power_mgmt
2. west flash ~/up_squared.sh
3. see error

**Logs and console output**
 Booting Zephyr OS build zephyr-v2.4.0-1137-g5803ee1e8183 
Running test suite power_management_test

START - test_power_idle
give way to idle thread
PASS - test_power_idle

**START - test_power_state_trans**

**it will stop at there, and fail with timeout.**

**Environment (please complete the following information):**
 - OS: Fedora31
 - Toolchain: zephyr-sdk-0.11.4
 - Commit ID: 5803ee1e8183

jenmwms, 2020-11-30:
This is still an issue asof `c08ef38`. Though similar symptom, seems different approach needed than #30275. `test_power_state_trans` frozen/still running after 9 minutes (stopwatch).

jenmwms, 2020-12-10:
I need to do a bisect to see where this started occurring.

nashif, 2020-12-11:
some issues with this test were fixed in cf8177406929f664b68c1f7b2071e4f87f209f97, any other issues most likely due to now https://github.com/zephyrproject-rtos/zephyr/issues/30573

closed 2020-12-11 by nashif